warpme/miniarch #73: How to Install MiniArch linux-aarch64-headers

JFLim1, 2025-01-07:
Hi @warpme,

Trying to install WiFi modules on MiniArch-EOS Orange Pi 5 Plus.  But noticed that MiniArch Kernel linux-aarch64-headers is not installed and is not sure how to install MiniArch linux-aarch64-headers-6.12.8.

warpme, 2025-01-09:
Well - you should have kernel headers already installed (my building process doing this at image creation pahase)
In done test on vanilla sd card downloaded form my site and booted on my opi5-plus:

```
[root@alarm alarm]# pacman -Qs linux
local/archlinux-keyring 20230225-1
    Arch Linux PGP keyring
local/archlinuxarm-keyring 20140119-2 (base-devel)
    Arch Linux ARM PGP keyring
local/base 3-1
    Minimal package set to define a basic Arch Linux installation
local/filesystem 2023.01.31-1
    Base Arch Linux files
local/iptables 1:1.8.9-1
    Linux kernel packet control tool (using legacy interface)
local/keyutils 1.6.3-1
    Linux Key Management Utilities
local/kmod 30-3
    Linux kernel module management tools and library
local/libcap-ng 0.8.3-1
    A library for Linux that makes using posix capabilities easy
local/linux-aarch64 6.12-1
    The Linux Kernel and modules - AArch64 multi-platform
local/linux-aarch64-api-headers 6.12-1

local/linux-api-headers 6.1.9-1
    Kernel headers sanitized for use in userspace
local/linux-firmware 20240709-1
    Firmware files for Linux
local/linux-firmware-whence 20230210.bf4115c-1
    Firmware files for Linux - contains the WHENCE license file which documents the vendor license details
local/net-tools 2.10-2
    Configuration tools for Linux networking
local/pacman-mirrorlist 20230206-1 (base)
    Arch Linux ARM mirror list for use by pacman
local/util-linux 2.38.1-1
    Miscellaneous system utilities for Linux
local/util-linux-libs 2.38.1-1
    util-linux runtime libraries
```
As you see there is package:  linux-aarch64-api-headers

Then doing: `ls /usr/include/linux*`

gives me:
```
[root@alarm alarm]# ls /usr/include/linux*
a.out.h		   cec.h		 futex.h	      ioam6_iptunnel.h	   mpls.h		personality.h	   serial_reg.h		user_events.h
acct.h		   cfm_bridge.h		 gameport.h	      ioctl.h		   mpls_iptunnel.h	pfkeyv2.h	   serio.h		userfaultfd.h
acrn.h		   cgroupstats.h	 gen_stats.h	      iommu.h		   mptcp.h		pfrut.h		   sev-guest.h		userio.h
adb.h		   chio.h		 genetlink.h	      iommufd.h		   mptcp_pm.h		pg.h		   shm.h		utime.h
adfs_fs.h	   cifs			 genwqe		      ioprio.h		   mqueue.h		phantom.h	   signal.h		utsname.h
affs_hardblocks.h  close_range.h	 gfs2_ondisk.h	      ip.h		   mroute.h		phonet.h	   signalfd.h		uuid.h
agpgart.h	   cm4000_cs.h		 gpio.h		      ip6_tunnel.h	   mroute6.h		pidfd.h		   smc.h		uvcvideo.h
aio_abi.h	   cn_proc.h		 gsmmux.h	      ip_vs.h		   mrp_bridge.h		pkt_cls.h	   smc_diag.h		v4l2-common.h
am437x-vpfe.h	   coda.h		 gtp.h		      ipc.h		   msdos_fs.h		pkt_sched.h	   smiapp.h		v4l2-controls.h
amt.h		   coff.h		 handshake.h	      ipmi.h		   msg.h		pktcdvd.h	   snmp.h		v4l2-dv-timings.h
android		   comedi.h		 hash_info.h	      ipmi_bmc.h	   mtio.h		pmu.h		   sock_diag.h		v4l2-mediabus.h
apm_bios.h	   connector.h		 hdlc		      ipmi_msgdefs.h	   nbd-netlink.h	poll.h		   socket.h		v4l2-subdev.h
arcfb.h		   const.h		 hdlc.h		      ipmi_ssif_bmc.h	   nbd.h		posix_acl.h	   sockios.h		vbox_err.h
arm_sdei.h	   coresight-stm.h	 hdlcdrv.h	      ipsec.h		   ncsi.h		posix_acl_xattr.h  sonet.h		vbox_vmmdev_types.h
aspeed-lpc-ctrl.h  counter.h		 hdreg.h	      ipv6.h		   ndctl.h		posix_types.h	   sonypi.h		vboxguest.h
aspeed-p2a-ctrl.h  cramfs_fs.h		 hid.h		      ipv6_route.h	   neighbour.h		ppdev.h		   sound.h		vdpa.h
aspeed-video.h	   cryptouser.h		 hiddev.h	      irqnr.h		   net.h		ppp-comp.h	   soundcard.h		vduse.h
atalk.h		   cuda.h		 hidraw.h	      isdn		   net_dropmon.h	ppp-ioctl.h	   spi			version.h
atm.h		   cxl_mem.h		 hpet.h		      iso_fs.h		   net_namespace.h	ppp_defs.h	   stat.h		vesa.h
atm_eni.h	   cyclades.h		 hsi		      isst_if.h		   net_tstamp.h		pps.h		   stddef.h		veth.h
atm_he.h	   cycx_cfm.h		 hsr_netlink.h	      ivtv.h		   netconf.h		pr.h		   stm.h		vfio.h
atm_idt77105.h	   dcbnl.h		 hw_breakpoint.h      ivtvfb.h		   netdev.h		prctl.h		   string.h		vfio_ccw.h
atm_nicstar.h	   dccp.h		 hyperv.h	      jffs2.h		   netdevice.h		psample.h	   sunrpc		vfio_zdev.h
[... 29 lines cut ...]
blkzoned.h	   f2fs.h		 if_pppol2tp.h	      map_to_14segment.h   nl80211-vnd-intel.h	rpmsg_types.h	   tls.h		virtio_vsock.h
bpf.h		   fadvise.h		 if_pppox.h	      map_to_7segment.h    nl80211.h		rseq.h		   toshiba.h		vm_sockets.h
bpf_common.h	   falloc.h		 if_slip.h	      matroxfb.h	   npcm-video.h		rtc.h		   tps6594_pfsm.h	vm_sockets_diag.h
bpf_perf_event.h   fanotify.h		 if_team.h	      max2175.h		   nsfs.h		rtnetlink.h	   trace_mmap.h		vmcore.h
bpfilter.h	   fb.h			 if_tun.h	      mctp.h		   nsm.h		rxrpc.h		   tty.h		vsockmon.h
bpqether.h	   fcntl.h		 if_tunnel.h	      mdio.h		   ntsync.h		scc.h		   tty_flags.h		vt.h
bsg.h		   fd.h			 if_vlan.h	      media		   nubus.h		sched		   types.h		vtpm_proxy.h
bt-bmc.h	   fdreg.h		 if_x25.h	      media-bus-format.h   nvme_ioctl.h		sched.h		   ublk_cmd.h		wait.h
btf.h		   fib_rules.h		 if_xdp.h	      media.h		   nvram.h		scif_ioctl.h	   udf_fs_i.h		watch_queue.h
btrfs.h		   fiemap.h		 ife.h		      mei.h		   omap3isp.h		screen_info.h	   udmabuf.h		watchdog.h
btrfs_tree.h	   filter.h		 igmp.h		      mei_uuid.h	   omapfb.h		sctp.h		   udp.h		wireguard.h
byteorder	   firewire-cdev.h	 iio		      membarrier.h	   oom.h		seccomp.h	   uhid.h		wireless.h
cachefiles.h	   firewire-constants.h  ila.h		      memfd.h		   openat2.h		securebits.h	   uinput.h		wmi.h
caif		   fou.h		 in.h		      mempolicy.h	   openvswitch.h	sed-opal.h	   uio.h		wwan.h
can		   fpga-dfl.h		 in6.h		      meye.h		   packet_diag.h	seg6.h		   uleds.h		x25.h
[... cut ...]
cciss_defs.h	   fsl_hypervisor.h	 input-event-codes.h  mman.h		   patchkey.h		seg6_local.h	   unistd.h		xilinx-v4l2-controls.h
cciss_ioctl.h	   fsl_mc.h		 input.h	      mmc		   pci.h		selinux_netlink.h  unix_diag.h		zorro.h
ccs.h		   fsmap.h		 io_uring.h	      mmtimer.h		   pci_regs.h		sem.h		   usb			zorro_ids.h
cdrom.h		   fsverity.h		 ioam6.h	      module.h		   pcitest.h		serial.h	   usbdevice_fs.h
cec-funcs.h	   fuse.h		 ioam6_genl.h	      mount.h		   perf_event.h		serial_core.h	   usbip.h
```
So for me all looks ok?

JFLim1, 2025-01-09:
Thanks. I do notice linux-aarch64-api-headers is installed.  Is linux-aarch64-api-headers the same as linux-aarch64-headers?  

After system updates the currentl kernel versiion is linux-aarch64-6.12.8-1 and linux-aarch64-api-headers-6.12.8-1 but linux-aarch64-headers as I can tell is not installed.  Was trying to install WiFi driver "rtw89-dkms-git".  Was told it needed the corresponding kernel-headers "linux-aarch64-headers-6.12.8-1"

warpme, 2025-01-09:
> Thanks. I do notice linux-aarch64-api-headers is installed. Is linux-aarch64-api-headers the same as linux-aarch64-headers?

No. linux-aarch64-api-headers is just for applications. If you want to build kernel modules you need inux-aarch64-headers.
Indeed - vanilla sd card image has only api-headers (for saving image size).
I updated miniarch script in c32e8160820c8e30d20c2e651c5941c2e81f490c to install also headers the user runs start 5  
Pls do:

1. run `start 5`
2. skip reboot
3. run again `start 5`
4. reboot

you should have now installed headers good to run dkms...

JFLim1, 2025-01-10:
> I updated miniarch script in c32e8160820c8e30d20c2e651c5941c2e81f490c to install also headers the user runs start 5 Pls do:
> 
> 1. run `start 5`

Tried for the last 1 hour.  Return error.

```
    Please confirm updating MiniArch speciffic components by pressing 'y'
    or press Enter (or any other key) to cancel ...
    (action may require root password)
 
y
:: Synchronizing package databases...
 miniarch.db failed to download
error: failed retrieving file 'miniarch.db' from miniarch.mooo.com : Connection timed out after 10001 milliseconds
error: failed to synchronize all databases (download library error)
 
    Requested command ended with error!
    Please fill GitHub ticket with details of this failed action. Thx in advance!
 
[alarm@MiniArch-opi5plus ~]$ 
```

JFLim1, 2025-01-10:
With the linux-aarch64-headers-6.12.9-1 installed on MiniArch-EOS proceeded to install "rtw89-dkms-git" but noticed there is an error:

```
==> Leaving fakeroot environment.
==> Finished making: rtw89-dkms-git r1117.9636e1b-1 (Fri 10 Jan 2025 04:38:32 PM +08)
==> Cleaning up...
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) rtw89-dkms-git-r1117.9636e1b-1

Total Installed Size:  8.46 MiB

:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                     [#######################] 100%
(1/1) checking package integrity                   [#######################] 100%
(1/1) loading package files                        [#######################] 100%
(1/1) checking for file conflicts                  [#######################] 100%
(1/1) checking available disk space                [#######################] 100%
:: Processing package changes...
(1/1) installing rtw89-dkms-git                    [#######################] 100%
:: Running post-transaction hooks...
(1/3) Arming ConditionNeedsUpdate...
(2/3) Install DKMS modules
==> dkms install --no-depmod rtw89/r1117.9636e1b -k 6.12.9

Error! Bad return status for module build on kernel: 6.12.9 (aarch64)
Consult /var/lib/dkms/rtw89/r1117.9636e1b/build/make.log for more information.
==> WARNING: `dkms install --no-depmod rtw89/r1117.9636e1b -k 6.12.9' exited 10
(3/3) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k 6.12.9 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
==> Starting build: '6.12.9'
```
/var/lib/dkms/rtw89/r1117.9636e1b/build/make.log:

```
DKMS make.log for rtw89/r1117.9636e1b for kernel 6.12.9 (aarch64)
Fri Jan 10 04:38:39 PM +08 2025
make -j`nproc` -C /lib/modules/6.12.9/build M=$PWD modules
make[1]: Entering directory '/usr/lib/modules/6.12.9/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-minimyth-linux-gnu-gcc (GCC) 14.2.0
  You are using:           gcc (GCC) 14.1.1 20240507
  CC [M]  /var/lib/dkms/rtw89/r1117.9636e1b/build/core.o
  CC [M]  /var/lib/dkms/rtw89/r1117.9636e1b/build/mac80211.o
  CC [M]  /var/lib/dkms/rtw89/r1117.9636e1b/build/mac.o
  CC [M]  /var/lib/dkms/rtw89/r1117.9636e1b/build/mac_be.o
  CC [M]  /var/lib/dkms/rtw89/r1117.9636e1b/build/phy.o
  CC [M]  /var/lib/dkms/rtw89/r1117.9636e1b/build/phy_be.o
  CC [M]  /var/lib/dkms/rtw89/r1117.9636e1b/build/cam.o
  CC [M]  /var/lib/dkms/rtw89/r1117.9636e1b/build/fw.o
/bin/sh: line 1: scripts/basic/fixdep: cannot execute binary file: Exec format error
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/rtw89/r1117.9636e1b/build/phy_be.o] Error 126
make[3]: *** Deleting file '/var/lib/dkms/rtw89/r1117.9636e1b/build/phy_be.o'
make[3]: *** Waiting for unfinished jobs....
/bin/sh: line 1: scripts/basic/fixdep: cannot execute binary file: Exec format error
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/rtw89/r1117.9636e1b/build/mac80211.o] Error 126
make[3]: *** Deleting file '/var/lib/dkms/rtw89/r1117.9636e1b/build/mac80211.o'
/bin/sh: line 1: scripts/basic/fixdep: cannot execute binary file: Exec format error
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/rtw89/r1117.9636e1b/build/mac_be.o] Error 126
make[3]: *** Deleting file '/var/lib/dkms/rtw89/r1117.9636e1b/build/mac_be.o'
/bin/sh: line 1: scripts/basic/fixdep: cannot execute binary file: Exec format error
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/rtw89/r1117.9636e1b/build/cam.o] Error 126
make[3]: *** Deleting file '/var/lib/dkms/rtw89/r1117.9636e1b/build/cam.o'
/bin/sh: line 1: scripts/basic/fixdep: cannot execute binary file: Exec format error
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/rtw89/r1117.9636e1b/build/mac.o] Error 126
make[3]: *** Deleting file '/var/lib/dkms/rtw89/r1117.9636e1b/build/mac.o'
/bin/sh: line 1: scripts/basic/fixdep: cannot execute binary file: Exec format error
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/rtw89/r1117.9636e1b/build/core.o] Error 126
make[3]: *** Deleting file '/var/lib/dkms/rtw89/r1117.9636e1b/build/core.o'
/bin/sh: line 1: scripts/basic/fixdep: cannot execute binary file: Exec format error
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/rtw89/r1117.9636e1b/build/phy.o] Error 126
make[3]: *** Deleting file '/var/lib/dkms/rtw89/r1117.9636e1b/build/phy.o'
/bin/sh: line 1: scripts/basic/fixdep: cannot execute binary file: Exec format error
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/rtw89/r1117.9636e1b/build/fw.o] Error 126
make[3]: *** Deleting file '/var/lib/dkms/rtw89/r1117.9636e1b/build/fw.o'
make[2]: *** [/usr/lib/modules/6.12.9/build/Makefile:1937: /var/lib/dkms/rtw89/r1117.9636e1b/build] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.12.9/build'
make: *** [Makefile:95: modules] Error 2
```
Not sure what it is related to this error.

warpme, 2025-01-10:
Ah - this is because kernel build scripts are built on mine x86 host (i'm cross-compiling).
They are included in `kernel-headers` package and yours dkms tries to use them - but on aarch64 machine (you are building natively).
It looks like linux kernel building infra is not prepared for cross-compile linux headers. Solving this is a bit out of mine project :-( 
huh....

Probably simples way will be to:
-download 6.12.9 kernel source code to your opi,
-run `make modules_prepare` to construct complete /scripts, then move it into `/lib/modules/`uname -r\`/build`

JFLim1, 2025-01-10:
> Ah - this is because kernel build scripts are built on mine x86 host (i'm cross-compiling).
> They are included in `kernel-headers` package and yours dkms tries to use them - but on aarch64 machine [...]

Thanks for clarifying and the info.

> Probably simples way will be to:
> -download 6.12.9 kernel source code to your opi,
> -run `make modules_prepare` to construct complete /scripts, then move it into `/lib/modules/`uname -r`/build`

A bit of a greenhorn in kernel compiling and is currently out of my depth.  Hopefully in future you might consider to include "rtw89-8852be" wifi driver in your kernel.

Thank you for your great works.

warpme, 2025-01-10:
I update kernel and it should now include 8852be module.
Can you update and try?

JFLim1, 2025-01-11:
> I update kernel and it should now include 8852be module.
> Can you update and try?

Thank you.  Have been trying.  Could not connect currently.  Will test again in 1 hrs time again.

```
    Please confirm updating MiniArch speciffic components by pressing 'y'
    or press Enter (or any other key) to cancel ...
    (action may require root password)
 
y
:: Synchronizing package databases...
 miniarch.db failed to download
error: failed retrieving file 'miniarch.db' from miniarch.mooo.com : Connection timed out after 10002 milliseconds
error: failed to synchronize all databases (download library error)
 
    Requested command ended with error!
    Please fill GitHub ticket with details of this failed action. Thx in advance!
 
[alarm@MiniArch-opi5plus ~]$
```

JFLim1, 2025-01-11:
> I update kernel and it should now include 8852be module.
> Can you update and try?

Somehow had the same build "rtw89-dkms-git" error.
```
sudo start 5
y
:: Synchronizing package databases...
 miniarch             1996.0   B  2.73 KiB/s 00:01 [#######################] 100%
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...

Packages (3) linux-aarch64-6.12.9-2  linux-aarch64-api-headers-6.12.9-2
             linux-aarch64-headers-6.12.9-2

Total Download Size:    69.43 MiB
Total Installed Size:  216.14 MiB
Net Upgrade Size:        0.62 MiB

:: Proceed with installation? [Y/n] Y
:: Retrieving packages...
 linux-aarch64-6....    39.8 MiB  1933 KiB/s 00:21 [#######################] 100%
 linux-aarch64-he...    28.1 MiB  2.03 MiB/s 00:14 [#######################] 100%
 linux-aarch64-ap...  1632.4 KiB  1174 KiB/s 00:01 [#######################] 100%
 Total (3/3)            69.4 MiB  1931 KiB/s 00:37 [#######################] 100%
(3/3) checking keys in keyring                     [#######################] 100%
(3/3) checking package integrity                   [#######################] 100%
(3/3) loading package files                        [#######################] 100%
(3/3) checking for file conflicts                  [#######################] 100%
(3/3) checking available disk space                [#######################] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
:: Processing package changes...
(1/3) upgrading linux-aarch64-headers              [#######################] 100%
(2/3) upgrading linux-aarch64                      [#######################] 100%
(3/3) upgrading linux-aarch64-api-headers          [#######################] 100%
:: Running post-transaction hooks...
(1/6) Arming ConditionNeedsUpdate...
(2/6) Updating module dependencies...
(3/6) Updating linux-aarch64 module dependencies...
(4/6) Install DKMS modules
==> dkms install --no-depmod rtw89/r1117.9636e1b -k 6.12.9

Error! Bad return status for module build on kernel: 6.12.9 (aarch64)
Consult /var/lib/dkms/rtw89/r1117.9636e1b/build/make.log for more information.
==> WARNING: `dkms install --no-depmod rtw89/r1117.9636e1b -k 6.12.9' exited 10
(5/6) Updating linux-aarch64 initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k 6.12.9 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
==> Starting build: '6.12.9'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [microcode]
==> WARNING: architecture 'aarch64' not supported, skipping hook
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
==> WARNING: No module containing the symbol 'drm_privacy_screen_register' found in: 'drivers/platform'
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux.img'
==> Initcpio image generation successful
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64.preset: 'fallback'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k 6.12.9 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-fallback.img -S autodetect
==> Starting build: '6.12.9'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [microcode]
==> WARNING: architecture 'aarch64' not supported, skipping hook
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
==> WARNING: No module containing the symbol 'drm_privacy_screen_register' found in: 'drivers/platform'
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
==> WARNING: Possibly missing firmware for module: 'ums_eneub6250'
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux-fallback.img'
==> Initcpio image generation successful
(6/6) Checking which packages need to be rebuilt
 
    System reboot is required
```

This might be a dumb question, if you have include the 8852be module in the kernel why it trying to install it using dkms?:

==> dkms install --no-depmod rtw89/r1117.9636e1b -k 6.12.9

warpme, 2025-01-11:
it looks issue is here:

```
:: Running post-transaction hooks...
(1/6) Arming ConditionNeedsUpdate...
(2/6) Updating module dependencies...
(3/6) Updating linux-aarch64 module dependencies...
(4/6) Install DKMS modules
==> dkms install --no-depmod rtw89/r1117.9636e1b -k 6.12.9
```
Your previous attempt to compile rtw89 left your system  in state where it still tries to compile rtw89
I don't have epierience with current dkms on Arch so I suggest search  Internet about: how to remove dkms building from post-transaction hooks

JFLim1, 2025-01-11:
> Your previous attempt to compile rtw89 left your system in state where it still tries to compile rtw89
> I don't have epierience with current dkms on Arch so I suggest search Internet about: how to remove dkms building from post-transaction hooks

Thank you.  Remove "rtw89-dkms-git" using "yay -R rtw89-dkms-git".  Not sure whether it remove the hook. 

Reboot still did not see WiFi available.

Do I have to reinstall linux-aarch64-6.12.9-2 and linux-aarch64-headers-6.12.9-2 from /var/cache/pacman/pkg?

Noticed /lib/firmware/rtw89 not available.

JFLim1, 2025-01-11:
Hi @warpme,

Resolved the WiFi 8852be driver issue on new kernel-6.12.9-2 by copying /lib/firmware/rtw89 drivers from another image to MiniArch-EOS /lib/firmware/.  With the /lib/firmware/rtw89 in place, WiFi now works  MiniArch-EOS-kernel-6.12.9-2 on Opi5-Plus.  Thank you.

No idea why /lib/firmware/rtw89 was missing eventhough "lsmod |grep rtw89" return rtw89_8852be is available. 

```
[alarm@MiniArch-opi5plus ~]$ lsmod | grep rtw89
rtw89_8852be           12288  0
rtw89_8852b           323584  1 rtw89_8852be
rtw89_8852b_common     61440  1 rtw89_8852b
rtw89_pci              73728  1 rtw89_8852be
rtw89_core            577536  3 rtw89_8852b,rtw89_pci,rtw89_8852b_common
mac80211             1236992  2 rtw89_core,rtw89_pci
cfg80211             1056768  3 rtw89_core,mac80211,rtw89_8852b_common
rfkill                 32768  8 rtw89_core,rfkill_gpio,bluetooth,cfg80211
[alarm@MiniArch-opi5plus ~]$ 
```