warpme/miniarch #74: Cannot Install MythTV on MiniArch - 40r Not Found

JFLim1, 2025-01-09:
Tried to install MythTV on MiniArch following guides:
https://github.com/warpme/miniarch/wiki/Installing-and-Setup-MythTV-on-MiniArch
```

[root@opi5plus ~]# wget http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
--2025-01-09 15:29:00--  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Resolving minimyth2.mooo.com (minimyth2.mooo.com)... 89.73.96.21
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-01-09 15:29:00 ERROR 404: Not Found.
```

warpme, 2025-01-09:
Well - this tooks to me like temporarily Internet hickup.
I just checked my and now access works ok.
may you try again and report?

JFLim1, 2025-01-10:
> may you try again and report?

Same issue.

```
[alarm@MiniArch-opi5plus ~]$ wget http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
--2025-01-10 23:26:50--  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Resolving minimyth2.mooo.com (minimyth2.mooo.com)... 89.73.96.21
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-01-10 23:26:51 ERROR 404: Not Found.
```

Will try again later or tomorrow.  May be is intermittent. 

"sudo start 5 " seems to be able to connect but NOT "wget http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz"

```
 sudo start 5

    Please confirm updating MiniArch speciffic components by pressing 'y'
    or press Enter (or any other key) to cancel ...
    (action may require root password)
 
y
:: Synchronizing package databases...
 miniarch is up to date
warning: linux-aarch64-headers-6.12.9-1 is up to date -- skipping
:: Starting full system upgrade...
 there is nothing to do
 
    System reboot is required!
 
    Press 'r' to reboot or
    Enter (or any other key) to continue ...
 

[alarm@MiniArch-opi5plus ~]$
``` 

Thanks.

JFLim1, 2025-01-11:
FYI, Trying this morning, the same issue except now it automatically will retry when it fails.  Now into 20 retries and doesn't seem to be able to download "mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz". 

Tried many times yesterday at different timing all failed.

```
[alarm@MiniArch-opi5plus ~]$ wget http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
--2025-01-11 09:46:21--  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Resolving minimyth2.mooo.com (minimyth2.mooo.com)... 89.73.96.21
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... failed: Connection timed out.
Retrying.

--2025-01-11 09:48:37--  (try: 2)  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... failed: Connection timed out.
Retrying.

--2025-01-11 09:50:52--  (try: 3)  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... failed: Connection timed out.
Retrying.

--2025-01-11 09:53:08--  (try: 4)  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... failed: Connection timed out.
Retrying.

--2025-01-11 09:55:26--  (try: 5)  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... failed: Connection timed out.
Retrying.

--2025-01-11 09:57:45--  (try: 6)  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... 

--2025-01-11 10:31:03--  (try:20)  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... failed: Connection timed out.
Giving up.

[alarm@MiniArch-opi5plus ~]$ 

```

warpme, 2025-01-11:
As it looks like my Internet provider is highly unreliable I changed miniarch packages hosting to github...

May you do: 
-execute
`sed -i 's|http://miniarch.mooo.com/pkg/$arch/miniarch|https://github.com/warpme/miniarch/raw/refs/heads/master|g' $(find . -name 'pacman-miniarch.conf') `
-run `start 5`

warpme, 2025-01-11:
Ah sorry!
So
in file: `/etc/pacman-miniarch.conf`
may you replace:
`http://miniarch.mooo.com/pkg/$arch/miniarch`
to
`https://github.com/warpme/miniarch/raw/refs/heads/master`

JFLim1, 2025-01-12:
> may you replace:
> `http://miniarch.mooo.com/pkg/$arch/miniarch`
> to
> `https://github.com/warpme/miniarch/raw/refs/heads/master`
> and try `start 5`

Ok edited /etc/pacman-miniarch.conf to:

```
[miniarch]
# Server = http://miniarch.mooo.com/pkg/$arch/miniarch
 Server = https://github.com/warpme/miniarch/raw/refs/heads/master
```

```
sudo start 5
y
:: Synchronizing package databases...
 miniarch             1865.0   B  2.15 KiB/s 00:01 [#######################] 100%
warning: linux-aarch64-headers-6.12.9-2 is up to date -- skipping
:: Starting full system upgrade...
 there is nothing to do
 
    System reboot is required!
 
    Press 'r' to reboot or
    Enter (or any other key) to continue ...
 

[alarm@MiniArch-opi5plus ~]$
``` 
Still cannot download/wget "mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz" result in "404 Not Found".

```
[alarm@MiniArch-opi5plus ~]$ wget http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
--2025-01-12 02:19:18--  http://minimyth2.mooo.com/mythtv-base-master.20241020.4e267985-1-aarch64.pkg.tar.xz
Resolving minimyth2.mooo.com (minimyth2.mooo.com)... 89.73.96.21
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-01-12 02:19:19 ERROR 404: Not Found.

[alarm@MiniArch-opi5plus ~]$
```
It seems to be able to connect but file Not Found? 
Just curious is it the file/package name correct?

JFLim1, 2025-01-13:
Hi @warpme,
Tried to wget the new mythtv-base-master.20250110.42970b8f-1-aarch64.pkg.tar.xz but also failed.  Connected but file 404 Not Found

```
[alarm@MiniArch-opi5plus ~]$ wget http://minimyth2.mooo.com/mythtv-base-master.20250110.42970b8f-1-aarch64.pkg.tar.xz
--2025-01-13 16:24:34--  http://minimyth2.mooo.com/mythtv-base-master.20250110.42970b8f-1-aarch64.pkg.tar.xz
Resolving minimyth2.mooo.com (minimyth2.mooo.com)... 89.73.96.21
Connecting to minimyth2.mooo.com (minimyth2.mooo.com)|89.73.96.21|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-01-13 16:24:35 ERROR 404: Not Found.

```

warpme, 2025-01-13:
I see you are tried to use old url (minimyth.mooo.com)
I switched to github so pls tray with new url (https://github.com/warpme/miniarch/raw/refs/heads/master)
(if you have new url in `/etc/pacman-miniarch.conf` you can try with):
`pacman --needed --overwrite \* --config /etc/pacman-miniarch.conf -Syu mythtv-base`

JFLim1, 2025-01-13:
> I switched to github so pls tray with new url (https://github.com/warpme/miniarch/raw/refs/heads/master)
> (if you have new url in `/etc/pacman-miniarch.conf` you can try with):

Yes, already switch to the new url yesterday.  The above command resulted in dependecies issue.  How to go about resolving to install MythTV on MiniArch-EOS?

[```
alarm@MiniArch-opi5plus ~]$ sudo pacman --needed --overwrite \* --config /etc/pacman-miniarch.conf -Syu mythtv-base
[sudo] password for alarm: 
:: Synchronizing package databases...
 miniarch             1865.0   B  2.19 KiB/s 00:01 [#######################] 100%
:: Starting full system upgrade...
resolving dependencies...
warning: cannot resolve "libcec", a dependency of "mythtv-base"
warning: cannot resolve "libzip", a dependency of "mythtv-base"
warning: cannot resolve "perl-dbd-mysql", a dependency of "mythtv-base"
warning: cannot resolve "perl-io-socket-inet6", a dependency of "mythtv-base"
warning: cannot resolve "perl-libwww", a dependency of "mythtv-base"
warning: cannot resolve "perl-xml-simple", a dependency of "mythtv-base"
warning: cannot resolve "perl-xml-xpath", a dependency of "mythtv-base"
warning: cannot resolve "python-future", a dependency of "mythtv-base"
warning: cannot resolve "python-lxml", a dependency of "mythtv-base"
warning: cannot resolve "python-mysqlclient", a dependency of "mythtv-base"
warning: cannot resolve "python-simplejson", a dependency of "mythtv-base"
warning: cannot resolve "python-dateutil", a dependency of "mythtv-base"
warning: cannot resolve "qt5-script", a dependency of "mythtv-base"
warning: cannot resolve "mariadb", a dependency of "mythtv-base"
:: The following package cannot be upgraded due to unresolvable dependencies:
      mythtv-base

:: Do you want to skip the above package for this upgrade? [y/N]
```

JFLim1, 2025-01-14:
Hi @warpme,

After installing MythTV no proceeded to "3.init database" but has "Access denied..." at:

```
[alarm@MiniArch-opi5plus ~]$ sudo mariadb-tzinfo-to-sql /usr/share/zoneinfo | mariadb mysql
ERROR 1044 (42000): Access denied for user ''@'localhost' to database 'mysql'
[alarm@MiniArch-opi5plus ~]$ 

```
Need guidance how to proceed/resolve this "Access denied..." issue.

warpme, 2025-01-14:
Maybe this will might be helpful: https://github.com/warpme/miniarch/wiki/Installing-and-Setup-MythTV-on-MiniArch

JFLim1, 2025-01-14:
> Maybe this might be helpful: https://github.com/warpme/miniarch/wiki/Installing-and-Setup-MythTV-on-MiniArch

Was following this guide. But now using the latest MythTV version and new url as opposed to on the above wiki.

Will try again.

JFLim1, 2025-01-14:
Hi @warpme,
Followed the steps as per https://github.com/warpme/miniarch/wiki/Installing-and-Setup-MythTV-on-MiniArch

All the command is with added with "sudo" for root.  All when well until it the step:

```
[alarm@MiniArch-opi5plus ~]$ sudo mariadb -u root < /usr/share/mythtv/mc.sql
[sudo] password for alarm: 
[alarm@MiniArch-opi5plus ~]$ sudo mariadb-tzinfo-to-sql /usr/share/zoneinfo | mariadb mysql
ERROR 1044 (42000): Access denied for user ''@'localhost' to database 'mysql'
[alarm@MiniArch-opi5plus ~]$ 
```

Stuck at the same step ERROR 1044 (42000): Access denied for user ''@'localhost' to database 'mysql'

How to proceed to resolve this to move on to the next step?

warpme, 2025-01-14:
May you try:

setup initial MythTV database by: Start terminal **and run as root**:

```
mariadb -u root < /usr/share/mythtv/mc.sql

mariadb-tzinfo-to-sql /usr/share/zoneinfo | mariadb mysql
```

JFLim1, 2025-01-14:
> setup initial MythTV database by: Start terminal **and run as root**:

Hi @warpme,
Do mean "sudo mariadb -u root < /usr/share/mythtv/mc.sql" is not sufficient.  Need to

use "sudo su" in terminal then 

```
[root@MiniArch-opi5plus alarm]# mariadb -u root < /usr/share/mythtv/mc.sql
[root@MiniArch-opi5plus alarm]# mariadb-tzinfo-to-sql /usr/share/zoneinfo | mariadb mysql
Warning: Unable to load '/usr/share/zoneinfo/SECURITY' as time zone. Skipping it.
[root@MiniArch-opi5plus alarm]# 
[root@MiniArch-opi5plus alarm]# mariadb -uroot -proot
Welcome to the MariaDB monitor.  Commands end with ; or \g.
Your MariaDB connection id is 5
Server version: 11.6.2-MariaDB Arch Linux

Copyright (c) 2000, 2018, Oracle, MariaDB Corporation Ab and others.

Type 'help;' or '\h' for help. Type '\c' to clear the current input statement.

MariaDB [(none)]> GRANT ALL ON mythconverg.* TO 'mythtv'@'%' IDENTIFIED BY 'mythtv';
Query OK, 0 rows affected (0.007 sec)

MariaDB [(none)]> FLUSH PRIVILEGES;
Query OK, 0 rows affected (0.001 sec)

MariaDB [(none)]> quit;
Bye
[root@MiniArch-opi5plus alarm]# 

[root@MiniArch-opi5plus alarm]# systemctl start mythbackend
[root@MiniArch-opi5plus alarm]# systemctl status mythbackend
● mythbackend.service - MythtTV Backend Server
     Loaded: loaded (/etc/systemd/system/mythbackend.service; disabled; preset:>
     Active: active (running) since Tue 2025-01-14 20:51:51 +08; 11s ago
 Invocation: 9c05249060d84704ac560e18b50500b7
   Main PID: 1677 (mythbackend)
      Tasks: 17 (limit: 19083)
     Memory: 48.6M (peak: 50M)
        CPU: 200ms
     CGroup: /system.slice/mythbackend.service
             └─1677 /usr/bin/mythbackend --systemd-journal --quiet

Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: ***** Use http://<yourBack>
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: **************************>
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: HTTPPool: Using maximum 16>
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: SSL: SSL Host key file (/h>
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Listening on TCP 0.0.0.0:6>
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Listening on TCP [::]:6544
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Listening on TCP 0.0.0.0:6>
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Listening on TCP [::]:6744
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Bonjour: Error: -65537
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Bonjour: Failed to registe>
lines 8-21/21 (END)
        CPU: 200ms
     CGroup: /system.slice/mythbackend.service
             └─1677 /usr/bin/mythbackend --systemd-journal --quiet

Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: ***** Use http://<yourBackend>:6544 to perform setup              *****
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: ***********************************************************************
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: HTTPPool: Using maximum 16 threads
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: SSL: SSL Host key file (/home/mythtv/.mythtv/certificates/key.pem) does not exist or is not readable
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Listening on TCP 0.0.0.0:6544
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Listening on TCP [::]:6544
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Listening on TCP 0.0.0.0:6744
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Listening on TCP [::]:6744
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Bonjour: Error: -65537
Jan 14 20:51:52 MiniArch-opi5plus mythbackend[1677]: Bonjour: Failed to register service.

```
What is the <backend_ip>?  Not sure what to enter http://<backend_ip>:6544?

warpme, 2025-01-16:
Well - MythTV is set of 2 components: backend(server) and frontend(client)
As <backend_ip> pls use IP address of machine where you started mythbackend

warpme, 2025-01-16:
Can we close this ticket?

JFLim1, 2025-01-16:
Oops.  I thought MythTV was part of MiniArch.  My bad.

Thanks for your assistancee.  I wanted to setup MythTV was basically to have vpu hw acceleration on videos. Couldn;t get MythTV to work.

Preferred choice for video is mpv and chromum but currently could not get to have vpu hardware acceleration on either mpv or chromium_v131.  Heard chromium_v132 might have v4l2 support.

closed 2025-01-16 by JFLim1

warpme, 2025-01-16:
Oh - so you may try MiniMyth2 then.
Currently it has also Kodi included and Kodi works well for me me on opi5 :-)

JFLim1, 2025-01-26:
> Oh - so you may try MiniMyth2 then.
> Currently it has also Kodi included and Kodi works well for me me on opi5 :-)

Thanks for the suggestions.  Just checking, does the MiniMyth2 with Kodi able to stream Youtube out of the box?  Haven't have any experience on Kodi at all.  Have been using mpv and chromium to stream youtube videos.