warpme/miniarch #76: MiniArch-EOS - If No Ethernet Wire Is Connected (WiFi Only) During Boot Up - Date & Time Is Set As Per The Last System Shutdown

JFLim1, 2025-01-12:
Device: Orange Pi 5 Plus with RTL8852BE-CG WiFi Module

Noticed that if Ethernet is not connected and only WiFi is connected on boot up, the Date & Time is the last date & time you shutdown.

If during power up Ethernet wire is connected Date & Time is set correctly via the internet.  But if no Ethernet wire is connected during power up and only WiFi connection, the date & time is set as per the last system shutdown time.

warpme, 2025-01-16:
I think this issue is non MiniArch related and is
- hw/board RTC related or
- OS related (ArchLinux Arm)
pls consult ArchLinux wiki's how to set time sync on systems with only wifi based networking...

JFLim1, 2025-01-18:
Hi @warpme,

Found the workaround or solution on System Time Not Sync when boot up with WiFi only on MiniArch-EOS.

The two files in /etc/systemd/network/en.network and /etc/systemd/network/eth.network are causing MiniArch-EOS to only sync the System Time with wired ethernet.  Disable them or rename to something else did the trick.

Noticed that the archlinuxarm image that can sync System Time when boot up with WiFi connection only the /etc/system/network directory is empty meaning does not have en.network or eth.network. By renaming the these two files in MiniArch-EOS resolved the issue.

Thanks for your taking the time to assist and support on the issues that I have raise on MiniArch.  MiniArch-EOS with mainline kernel is working well on OPI5-Plus.  Looking for solution on Chromium and MPV having vpu hardware acceleration on MiniArch-EOS.

Thank you.

warpme, 2025-01-24:
closed with status: not miniarch issue

closed 2025-01-24 by warpme