warpme/miniarch #77: Start 5 return: error: target not found: linux-aarch64-headers

JFLim1, 2025-01-19:
Was just doing "Update MiniArch artefacts":

```
  (1) Init pacman keyring and populate packages signing keys
  (2) Full update (ArchLinuxArm codebase)
  (3) Install on eMMC (optinal)
  (4) Resize footfs partition
  (5) Update MiniArch artefacts
  (6) Install EndeavourOS
 
or press Enter to exit...

5
 
    Please confirm updating MiniArch speciffic components by pressing 'y'
    or press Enter (or any other key) to cancel ...
    (action may require root password)
 
y
:: Synchronizing package databases...
 miniarch                                 1668.0   B  2024   B/s 00:01 [#######################################] 100%
error: target not found: linux-aarch64-headers
 
    Requested command ended with error!
    Please fill GitHub ticket with details of this failed action. Thx in advance!
 
[alarm@MiniArch-opi5plus ~]$ 

```
FYI, currently on kernel-6.12.9-2.

warpme, 2025-01-20:
Ah - my bad. 
it was typo in one of my scripts. 
now should be ok :-)

JFLim1, 2025-01-20:
Thank you.  Upgraded to the new kernel-6.12.10-1 and kernel-headers  successfully.