warpme/miniarch #78: Memory quantity error / Installation on eMMC does not work

matheusullmann, 2025-01-22:
Hey guys!

I have the next TV Box card, commercial model MXQ Pro 4K 5G, with 2Gb of RAM. Board model IK316Q44_V1.0. 
The image "MiniArch-20240715-6.12.9-board-h313.x96_q_v5.1-SD-Image.img.xz" works perfectly, except for the amount of memory which says 1Gb.
- HDMI works.
- Rj45 works.
- Wi-fi not tested.
- Recording to internal memory does not work yet.
- Reports 1Gb of memory instead of 2Gb. The datasheet also states 2Gb for the K4B2G0446Q chip.

How can I make the system detect the correct amount of memory?
Is there anything I can do to fix the error in step 5 "select (5) to update MiniArch artefacts", could this be preventing the installation on the eMMC?

[image]

[image]

[image]

[image]

[image]

warpme, 2025-01-23:
Hi,
Thx for report.
I'm glad MiniArch works well for you :-)

One Q: on pictures i see 4 chips - each 2Gb
So 4 of them gives 8Gb=1GB.
Do i miss something here?

And regarding: _I can do to fix the error in step 5_
May you pls be a bit more specific here?

warpme, 2025-01-31:
closed due inactivity...

closed 2025-01-31 by warpme