warpme/miniarch #79: Porting Radxa DTBs overlays for the zero 3W?

gbin, 2025-01-24:
It looks like some entries are missing in the DTB shipped with miniarch.

For example I would like to use the Radxa Camera 8M and its dtb overlay looks like this:

```
sal ➜  overlays (main) ✗ cat  radxa-zero3-radxa-camera-8m.dts
/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/pinctrl/rockchip.h>

/ {
        metadata {
                title = "Enable Radxa Camera 8M 219";
                compatible = "radxa,zero3";
                category = "camera";
                exclusive = "csi2_dphy0";
                description = "Enable Radxa Camera 8M 219.";
        };
};

&{/} {
        clk_cam_24m: external-camera-clock-24m {
                status = "okay";
                compatible = "fixed-clock";
                clock-frequency = <24000000>;
                clock-output-names = "clk_cam_24m";
                #clock-cells = <0>;
        };

        camera_pwdn_gpio: camera-pwdn-gpio {
                status = "okay";
                compatible = "regulator-fixed";
                regulator-name = "camera_pwdn_gpio";
                regulator-always-on;
                regulator-boot-on;
                enable-active-high;
                gpio = <&gpio3 RK_PC6 GPIO_ACTIVE_HIGH>;
        };
};

&i2c2 {
        status = "okay";
        pinctrl-names = "default";
        pinctrl-0 = <&i2c2m1_xfer>;
        #address-cells = <1>;
        #size-cells = <0>;

        camera_imx219: camera-imx219@10 {
                status = "okay";
                compatible = "sony,imx219";
                reg = <0x10>;
                clocks = <&clk_cam_24m>;
                clock-names = "xvclk";
                rockchip,camera-module-index = <0>;
                rockchip,camera-module-facing = "back";
                rockchip,camera-module-name = "RADXA-CAMERA-8M";
                rockchip,camera-module-lens-name = "default";

                port {
                        ucam_out0: endpoint {
                                remote-endpoint = <&mipi_in_ucam0>;
                                data-lanes = <1 2>;
                        };
                };
        };
};

&csi2_dphy_hw {
        status = "okay";
};

&csi2_dphy0 {
        status = "okay";

        ports {
                #address-cells = <1>;
                #size-cells = <0>;

                port@0 {
                        reg = <0>;
                        #address-cells = <1>;
                        #size-cells = <0>;

                        mipi_in_ucam0: endpoint@1 {
                                reg = <1>;
                                remote-endpoint = <&ucam_out0>;
                                data-lanes = <1 2>;
                        };
                };

                port@1 {
                        reg = <1>;
                        #address-cells = <1>;
                        #size-cells = <0>;

                        dphy0_out: endpoint@1 {
                                reg = <1>;
                                remote-endpoint = <&isp0_in>;
                        };
                };
        };
};

&rkisp_vir0 {
        status = "okay";

        port {
                #address-cells = <1>;
                #size-cells = <0>;

                isp0_in: endpoint@0 {
                        reg = <0>;
                        remote-endpoint = <&dphy0_out>;
                };
        };
};

&rkisp {
        status = "okay";
};

&rkisp_mmu {
        status = "okay";
};

&rkcif_mmu {
        status = "okay";
};

&rkcif {
        status = "okay";
};
```

But all those ie. &i2c2 and rk refs are missing it looks like.
I tried to directly use their dtb+overlays on this kernel but unsurprisingly it doesn't work.

Maybe an option to use their supported kernel might help?

warpme, 2025-01-24:
Hi,
Let me look on overlays issue.

Re: using "their" kernel: well this will be against miniarch philosophy: single mainline kernel for all 40+ different devices....

gbin, 2025-01-24:
@warpme hopefully it is not too much of a rabbit hole for you, I lost 4 days on this :( but I think dtbs are not my thing lol

warpme, 2025-01-24:
@gbin 
Pls note: one thing is dt overlays support; other thing is given soc pheri. support....

Generally: supporting all soc pheri with mainline kernel is task for years (keeping in mind speed of adding new hw support by kernel guys). This is why bsp kernels had and will have long life.

I suspect many thing are missing for camera support in mainline kernel so for your usecase bst will be probably switch to distro running bsp kernel....

gbin, 2025-01-24:
@warpme I am trying that... Just discovered that they used a custom uboot, they load stuff from the main OS partition, this thing keeps on giving Oo.

warpme, 2025-02-03:
May we close this ticket with status: hard dependency on upstream (issue can't be resolved due missing  upstream kernel functionality)?

gbin, 2025-02-03:
Yes exactly I tried but it is intractable: the kernel + dtb is one. Tried to use your root fs with their kernel, it doesn't work neither.

So yeah stuck with Debian from 1982 Oo...

Thanks!

closed 2025-02-03 by gbin