warpme/miniarch #83: v20250206 OrangePi5 no Boot

StuartIanNaylor, 2025-02-21:
Dunno as would have to route out a serial console but the Debian Server Image from the OrangePi site does.

warpme, 2025-02-22:
Hi,
I'm a bit lost with this report...
What exactly issue you have?

btw:
i downloaded Opi5 image, flashed to sd card, inserted into opi5 and booted.
Uart gives me like this:

```
DDR Version V1.08 20220617
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x1 Samsung
CH0 RX Vref:35.7%, TX Vref:20.8%,20.8%
CH1 RX Vref:32.7%, TX Vref:21.8%,21.8%
CH2 RX Vref:33.7%, TX Vref:19.8%,19.8%
CH3 RX Vref:34.7%, TX Vref:20.8%,19.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-orangepi (Apr 23 2023 - 11:47:44)
Trying to boot from MMC1
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(7612223b82...) + OK
## Checking u-boot 0x00a00000 ... sha256(7cc579dc16...) + OK
## Checking fdt-1 0x00acf5b0 ... sha256(e112aba2ff...) + OK
## Checking atf-2 0x000f0000 ... sha256(b2af21b504...) + OK
## Checking atf-3 0xff100000 ... sha256(70505bb764...) + OK
Jumping to U-Boot(0x00a00000) via ARM Trusted Firmware(0x00040000)
Total: 689.938 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-868-g040d2de11:derrick.huang, fwver: v1.48
NOTICE:  BL31: Built : 15:02:44, Dec 19 2024
INFO:    spec: 0x13
INFO:    code: 0x88
INFO:    ext 32k is valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    l3 cache partition cfg-0
INFO:    system boots from cpu-hwid-0
INFO:    disable memory repair
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
ERROR:   dfs get fsp_params[0] error, 0xfead0001 != 0xfead0004
ERROR:   dfs get fsp_params[1] error, 0xa54 != 0xfead0004
ERROR:   dfs get fsp_params[2] error, 0xbc != 0xfead0004
ERROR:   dfs get fsp_params[3] error, 0x3f0033 != 0xfead0004
ERROR:   loader&trust unmatch!!! Please update loader if need enable dmc
ERROR:   current trust bl31 need match with loader ddr bin V1.13 or newer
ERROR:   current loader need match with trust bl31 V1.07-V1.37
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0xa00000
INFO:    SPSR = 0x3c9

U-Boot 2025.01 (Feb 06 2025 - 15:43:30 +0100)

Model: Xunlong Orange Pi 5
[... 185 lines cut ...]
[.[0;32m  OK  .[0m] Reached target .[0;1;39mBasic System.[0m.
         Starting .[0;1;39mD-Bus System Message Bus.[0m...
         Starting .[0;1;39mSSH Key Generation.[0m...
         Starting .[0;1;39mUser Login Management.[0m...
         Starting .[0;1;39mPermit User Sessions.[0m...
[.[0;32m  OK  .[0m] Finished .[0;1;39mPermit User Sessions.[0m.
[.[0;32m  OK  .[0m] Started .[0;1;39mGetty on tty1.[0m.
[.[0;32m  OK  .[0m] Started .[0;1;39mSerial Getty on ttyS2.[0m.
[.[0;32m  OK  .[0m] Reached target .[0;1;39mLogin Prompts.[0m.
[.[0;32m  OK  .[0m] Started .[0;1;39mD-Bus System Message Bus.[0m.
[.[0;32m  OK  .[0m] Started .[0;1;39mUser Login Management.[0m.
         Starting .[0;1;39mHostname Service.[0m...
[.[0;32m  OK  .[0m] Started .[0;1;39mVerify integrity of password and group files.[0m.
[.[0;32m  OK  .[0m] Finished .[0;1;39mSSH Key Generation.[0m.
[.[0;32m  OK  .[0m] Started .[0;1;39mHostname Service.[0m.
[.[0;32m  OK  .[0m] Started .[0;1;39mOpenSSH Daemon.[0m.
[.[0;32m  OK  .[0m] Reached target .[0;1;39mMulti-User System.[0m.
[.[0;32m  OK  .[0m] Reached target .[0;1;39mGraphical Interface.[0m.

Arch Linux 6.12.12 (ttyS2)

alarm login: 
```  

for me all this looks ok. Isn't?

StuartIanNaylor, 2025-02-22:
I dunno for some reason didn't boot for me, doesn't matter as getting rid of all my SBC and was just putting your image on as you do so many and the Opi5 was the only one it seemed to fail on.
Apols as they are going on ebay hence the brief report

closed 2025-02-22 by StuartIanNaylor