warpme/miniarch #85: AIC8800 drivers

gyscos, 2025-03-11:
Hi!

Currently running a radxa zero-3W with the latest kernel and firmware from this repo, however I get a lot of RX packet dropped on the wifi interface.

I was curious if the linux-firmware package here includes the latest changes from https://github.com/radxa-pkg/aic8800?

warpme, 2025-03-11:
Hi,
aic8800 driver code + firmware was not updated by vendor since 9 months.
This crappy code has a lot of issues and it is really shame radxa is using this junk as wifi on their so nice boards..... 
I contacted radxa many times about quality of aic8800 code. They - the same like me - are 100% dependent on aicsemi.
aicsemi seems totally deaf to any issue reports so I'm afraid not much can be done with this :-(

gyscos, 2025-03-13:
Thanks for the quick answer!

I do think the `linux-firmware` package from this repo has worse performance with AIC8800 than what radxa provides in the repo I linked above.

Running up-to-date miniarch, there must be some packet rejection issue because while I can ssh, `iperf3` reports a speed of 0.0 between the radxa-zero3w and another device on the network.
By comparison, in the same location, the "official" debian bullseye image with `aic8800-sdio-dkms` reports ~5Mb/s (this is still way less than a raspberry pi Zero W in the same location, which reports 50Mb/s).

I'll see if I can use the "official" deb dkms package in miniarch, and if it helps. I'm not familiar with overlays and device tree much though so I might mess things up :(

warpme, 2025-03-14:
_By comparison, in the same location, the "official" debian bullseye image with aic8800-sdio-dkms reports ~5-10Mb/s (this is still way less than a raspberry pi Zero W in the same location, which reports 40-50Mb/s, though maybe the raspberry pi has a better antenna)._

this is very interesting. 
May you pls provide me link to sources of this "official" package?
I can compare sources + fw - maybe there is some magic changes making debian code working better?

warpme, 2025-03-14:
_Is it possible to rebuild the linux-aarch64-headers package, but with aarch64 binaries instead of x86_64 ones?_

I'm already providing kernel-header package suitable (in theory) to build extra/out-of-tree kenrel modules.
Key issue is that kernel build scripts included in kernel-header package are built by kernel infra for x86 host.
Those binaries (included in kernel-headers package) dkms tries to use yours aarch64 machine (as you are building natively with yours dmks). As they are build for x86_64 (on/by my cross-compiling host) - yours dkms process fails....

All this shows to me like linux kernel building infra is not prepared to build correct linux headers when user (me) cross-compiling kernel-headers. 

Solving this is a bit beyond of mine project :-(

Probably simples way will be to:
-download kernel source code to your machine,
-run `make modules_prepare` to construct complete /scripts
-move them into `/lib/modules/uname -r`/build``

warpme, 2025-03-16:
I uploaded kernel packages with most recent radxa aic8800 fixes.
With this indeed i see improvement with transfers stability:

before (last colon is wifi 5G Mbps throughput):
```
131072  87380  87380    1.04       52.40
131072  87380  87380    1.11      129.59
131072  87380  87380    1.13       84.85
131072  87380  87380    1.00        0.00
131072  87380  87380    1.09       73.88
131072  87380  87380    1.09      122.56
131072  87380  87380    1.10      136.01
131072  87380  87380    1.17      142.33
131072  87380  87380    1.10      127.69
131072  87380  87380    1.12      119.76
131072  87380  87380    1.08       70.80
131072  87380  87380    1.20      118.08
131072  87380  87380    1.11      116.58
131072  87380  87380    1.13      114.24
131072  87380  87380    1.12      124.49
131072  87380  87380    1.07       61.91
131072  87380  87380    1.10      124.23
131072  87380  87380    1.14      106.83
131072  87380  87380    1.03       28.98
131072  87380  87380    1.07      101.39
131072  87380  87380    1.11       77.39
131072  87380  87380    1.19       20.00
131072  87380  87380    1.10      102.26
131072  87380  87380    1.08       63.25
131072  87380  87380    1.08       65.42
131072  87380  87380    1.23       99.36
131072  87380  87380    1.15       63.69
131072  87380  87380    1.31       28.58
131072  87380  87380    1.08      118.31
131072  87380  87380    1.16      100.30   
```

after:
```
131072  87380  87380    1.07      142.79
131072  87380  87380    1.12      172.30
131072  87380  87380    1.10      123.85
131072  87380  87380    1.09      108.20
131072  87380  87380    1.15      156.44
131072  87380  87380    1.08      114.60
131072  87380  87380    1.09      151.54
131072  87380  87380    1.10      169.80
131072  87380  87380    1.14      147.41
131072  87380  87380    1.09      157.08
131072  87380  87380    1.10      155.22
131072  87380  87380    1.10      166.01
131072  87380  87380    1.10      156.17
131072  87380  87380    1.07      167.89
131072  87380  87380    1.10      109.33
131072  87380  87380    1.16      135.97
131072  87380  87380    1.12      147.48
131072  87380  87380    1.09      150.73
131072  87380  87380    1.09      174.36
131072  87380  87380    1.10      151.21
131072  87380  87380    1.07      156.42
131072  87380  87380    1.09      165.90
131072  87380  87380    1.15      154.92
131072  87380  87380    1.07      163.48
131072  87380  87380    1.12      167.01
131072  87380  87380    1.08      167.43
131072  87380  87380    1.12      174.35
131072  87380  87380    1.11      164.38
131072  87380  87380    1.13      148.07
131072  87380  87380    1.10      145.60
131072  87380  87380    1.14      167.16
131072  87380  87380    1.14      155.89
131072  87380  87380    1.13      146.06
131072  87380  87380    1.07      139.43
131072  87380  87380    1.10      150.22
131072  87380  87380    1.10      155.48
131072  87380  87380    1.13      167.76
131072  87380  87380    1.12      162.42
131072  87380  87380    1.07      141.06
131072  87380  87380    1.09      126.93
131072  87380  87380    1.06      132.58
131072  87380  87380    1.04       76.55
131072  87380  87380    1.08      147.42
131072  87380  87380    1.10      132.23
131072  87380  87380    1.15      158.65
131072  87380  87380    1.16      163.89
131072  87380  87380    1.14      155.22
131072  87380  87380    1.15      129.36
131072  87380  87380    1.12      149.21
131072  87380  87380    1.13      143.60
131072  87380  87380    1.10      153.71
131072  87380  87380    1.18      154.74
131072  87380  87380    1.07       99.11
131072  87380  87380    1.09      156.57
131072  87380  87380    1.13      152.15
131072  87380  87380    1.08      159.73
131072  87380  87380    1.11      157.77
131072  87380  87380    1.11      162.45
131072  87380  87380    1.10      161.96
131072  87380  87380    1.19      167.08
131072  87380  87380    1.09      123.85
131072  87380  87380    1.14      124.78
131072  87380  87380    1.14      142.93
131072  87380  87380    1.10      140.13
131072  87380  87380    1.12      134.33
131072  87380  87380    1.16      126.91
131072  87380  87380    1.11      135.11
131072  87380  87380    1.14      137.53
131072  87380  87380    1.15      137.20
131072  87380  87380    1.14      157.91
131072  87380  87380    1.13      158.63
```

warpme, 2025-03-23:
Closing as fixed.

closed 2025-03-23 by warpme