warpme/miniarch #87: Cannot Upgrade to linux-aarch64-6.14-1 from linux-aarch64 6.14.rc6-1

JFLim1, 2025-03-26:
Tried to upgrade to the latest kernel linux-aarch64-6.14-1 from linux-aarch64 6.14.rc6-1 using "sudo start 5".

```
or press Enter to exit...

5
 
    Please confirm updating MiniArch speciffic components by pressing 'y'
    or press Enter (or any other key) to cancel ...
    (action may require root password)
 
y
:: Synchronizing package databases...
 miniarch                      1231.0   B  1572   B/s 00:01 [################################] 100%
warning: downgrading package linux-aarch64-headers (6.14.rc6-1 => 6.14-1)
:: Starting full system upgrade...
warning: linux-aarch64: local (6.14.rc6-1) is newer than miniarch (6.14-1)
warning: linux-aarch64-api-headers: local (6.14.rc6-1) is newer than miniarch (6.14-1)
resolving dependencies...
looking for conflicting packages...

Packages (1) linux-aarch64-headers-6.14-1

Total Download Size:    28.47 MiB
Total Installed Size:  100.98 MiB
Net Upgrade Size:        0.13 MiB

:: Proceed with installation? [Y/n] Y
```
Only linux-aarch64-headers-6.14-1 was upgraded.

warpme, 2025-03-28:
It looks like pacman sees 6.14-rc6 as newer to 6.14

You may try manually update kernel: 

`wget https://github.com/warpme/miniarch/raw/refs/heads/master/linux-aarch64-6.14-1-any.pkg.tar.gz`

then run in terminal as root:

`pacman -U linux-aarch64-6.14-1-any.pkg.tar.gz`

this is pacman issue.
in the future i will not offer anymore .rc versions of kernel...

JFLim1, 2025-03-30:
Thanks.

Managed to installed both linux-aarch64-6.14-1-any.pkg.tar.gz linux-aarch64-api-headers-6.14-1-any.pkg.tar.gz using 
wget https://github.com/warpme/miniarch/raw/refs/heads/master/linux-aarch64-6.14-1-any.pkg.tar.gz
wget https://github.com/warpme/miniarch/raw/refs/heads/master/linux-aarch64-api-headers-6.14-1-any.pkg.tar.gz

sudo pacman -Ulinux-aarch64-6.14-1-any.pkg.tar.gz linux-aarch64-api-headers6.14-1-any.pkg.tar.gz

closed 2025-03-30 by warpme

JFLim1, 2025-03-30:
If not mistaken previously the with rc the naming convention is like linux-aarch64-6.14-rc  instead of linux-aarch64.rc.