warpme/miniarch #88: Opi5-Plus HDMI1 No Audio/Sound Output - Kernel-6.14-1

JFLim1, 2025-04-13:
Hi @warpme,

Device: Opi5-Plus.  
Kernel: 6.14-1
Desktop Environment: Gnome-48
Not sure whether this is a known issue.  HDMI0 audio/sound is available but on HDMI1 there is no audio/sound output.

warpme, 2025-04-14:
Well - what:
`aplay -L`
`aplay -l`
tells you?

JFLim1, 2025-04-14:
```
[alarm@opi5plus ~]$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pipewire
    PipeWire Sound Server
pulse
    PulseAudio Sound Server
speex
    Plugin using Speex DSP (resample, agc, denoise, echo, dereverb)
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Default ALSA Output (currently PipeWire Media Server)
sysdefault:CARD=rockchiphdmi0
    rockchip-hdmi0, fddf0000.i2s-i2s-hifi i2s-hifi-0
    Default Audio Device
usbstream:CARD=rockchiphdmi0
    rockchip-hdmi0
    USB Stream Output
sysdefault:CARD=rockchiphdmi1
    rockchip-hdmi1, fddf4000.i2s-i2s-hifi i2s-hifi-0
    Default Audio Device
usbstream:CARD=rockchiphdmi1
    rockchip-hdmi1
    USB Stream Output
sysdefault:CARD=Analog
    Analog, fe470000.i2s-es8328-hifi-analog es8328-hifi-analog-0
    Default Audio Device
usbstream:CARD=Analog
    Analog
    USB Stream Output
```

```
[alarm@opi5plus ~]$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: rockchiphdmi0 [rockchip-hdmi0], device 0: fddf0000.i2s-i2s-hifi i2s-hifi-0 [fddf0000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: rockchiphdmi1 [rockchip-hdmi1], device 0: fddf4000.i2s-i2s-hifi i2s-hifi-0 [fddf4000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Analog [Analog], device 0: fe470000.i2s-es8328-hifi-analog es8328-hifi-analog-0 [fe470000.i2s-es8328-hifi-analog es8328-hifi-analog-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
[alarm@opi5plus ~]$ 
```

warpme, 2025-04-14:
I think issue is in EOS.
I just tested mm2 6.14.1 kernel on minimyth2 and hdmi1 audio works ok.
As miniarch uses the same kernel (even binaries are the same) - I don't believe your issue is miniarch related. (in sense miniarch code I'm developing)
May you test with some command line tools (e.g. alsa aplay / speaker-test)?

JFLim1, 2025-04-15:
> May you test with some command line tools (e.g. alsa aplay / speaker-test)?

Figure what you meant.

Select "HDMI1" audio output from Gnome Settings. Then run "speaker-test" command on Terminal.  

No Audio/Sound output from HDMI1 which is the TV.  HDMI0 which is the monitor has audio output.

```
[alarm@opi5plus ~]$ speaker-test

speaker-test 1.2.13

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 128 to 2097152
Period size range from 64 to 1048576
Periods = 4
was set period_size = 12000
was set buffer_size = 48000
 0 - Mono
Time per period = 1.964632
 0 - Mono
Time per period = 3.029397
 0 - Mono
Time per period = 2.986620
 0 - Mono
Time per period = 2.986675
 0 - Mono
Time per period = 3.029314
 0 - Mono
Time per period = 2.986500
 0 - Mono
Time per period = 2.986654
 0 - Mono
Time per period = 3.029236
 0 - Mono
Time per period = 2.986553
 0 - Mono
Time per period = 2.986643
 0 - Mono
Time per period = 2.986579
 0 - Mono
Time per period = 3.029271
 0 - Mono
Time per period = 2.986601
 0 - Mono
Time per period = 2.986613
 0 - Mono
Time per period = 3.029225
 0 - Mono
Time per period = 2.986653
 0 - Mono
Time per period = 2.986593
 0 - Mono
Time per period = 3.029260
 0 - Mono
Time per period = 2.986536
 0 - Mono
Time per period = 2.986632
 0 - Mono
Time per period = 3.029283
 0 - Mono
Time per period = 2.986632
 0 - Mono
Time per period = 2.986528
 0 - Mono
Time per period = 3.029259
 0 - Mono
Time per period = 2.986590
 0 - Mono
Time per period = 2.986589
 0 - Mono
Time per period = 2.986591
 0 - Mono
Time per period = 3.029210
 0 - Mono
Time per period = 2.986631
 0 - Mono
Time per period = 2.986586
 0 - Mono
Time per period = 3.029270
 0 - Mono
Time per period = 2.986574
 0 - Mono
Time per period = 2.986560
 0 - Mono
Time per period = 3.029253
 0 - Mono
Time per period = 2.986611
 0 - Mono
Time per period = 2.986567
 0 - Mono

```

JFLim1, 2025-04-22:
Hi @warpme,
FYI, Upgraded to kernel-6.14.3-1, still no audio/sound on HDMI1 on Opi5-Plus.

JFLim1, 2025-05-29:
Hi @warpme,

FYI, A feedback.

 Just upgraded MiniArch-EOS-Gnome (Opi5-Plus) to kernel-6.15-1, but unfortunate still no audio/sound outp on HDMI1.

warpme, 2025-06-01:
I done test with minimyht2 and audio plays ok on hdmi0 and hdmi1
I think issue is in EOS - not in miniarch.
As EOS is out of scope of miniarch - may we close this ticket with status "eos bug"?

JFLim1, 2025-06-02:
EOS meaning reference to Archlinux-Arm Gnome DE?

Am not familiar what causes NO hdmi1 audio output. 

Sure, close as per your suggestion.

Thanks.

warpme, 2025-06-02:
EOS is EndeavourOS :-)

JFLim1, 2025-07-15:
Hi @warpme,

Just an update on HDMI-1 Audio output.  Managed to get HDMI-1 Audio output on MiniArch-EOS-KDE by first login to Plasma-X11 and was able to select HDMI-1 Audio output.  Then switch to Plasma-Wayland and HDMI-1 Audio continue to be available.

On MiniArch-EOS-Gnome somehow switching to Gnome-X11 did not have the option to select HDMI-1 audio output.  Thus far not successful, may be I am missing something.