warpme/miniarch #89: Corrupted packages

scoreabasm, 2025-04-19:
Problem using start / step 2 (full update) with multiple invalid or corrupted package (PGP signature). Same problem with various mirrors. Please help. 

Details hereunder (15 packages in error):
error: glibc: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/glibc-2.41+r6+gcf88351b685d-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: gcc-libs: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/gcc-libs-14.2.1+r753+g1cd744a6828f-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: ncurses: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/ncurses-6.5-3-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: openssl: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/openssl-3.5.0-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: ca-certificates-mozilla: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/ca-certificates-mozilla-3.110-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: icu: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/icu-76.1-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: tpm2-tss: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/tpm2-tss-4.1.3-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: shadow: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/shadow-4.17.4-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: util-linux: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/util-linux-2.41-4-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: systemd: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/systemd-257.5-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: psmisc: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/psmisc-23.7-1-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
error: libnetfilter_conntrack: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is invalid
:: File /var/cache/pacman/pkg/libnetfilter_conntrack-1.0.9-2-aarch64.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).

warpme, 2025-04-20:
Hi,
I think your issue was related to temp. problems with archlinux arm repos in Internet.
I just:
-downloaded orangepi5 sd card image
-flashed it to sd card
-inserted card
-booed device
-done `start 1` 
-done `start 2`
all was ok (no any issues)

scoreabasm, 2025-04-20:
> Hi,
> I think your issue was related to temp. problems with archlinux arm repos in Internet.
> I just:
> -downloaded orangepi5 sd card image
> -flashed it to sd card
> -inserted card
> -booted device
> -done `start 1` 
> -done `start 2`
> all was ok (no any issues)
>  

Hi, ok I'll try again and report back asap. Many thanks.

scoreabasm, 2025-04-20:
> Hi, I think your issue was related to temp. problems with archlinux arm repos in Internet. I just: -downloaded orangepi5 sd card image -flashed it to sd card -inserted card -booted device -done `start 1` -done `start 2` all was ok (no any issues)

No sorry, downloaded image again, burned to a new sd card, booted and 'start 1' -done, then 'start 2' with same problem... Any other idea?

warpme, 2025-04-21:
may you try procedure from: https://github.com/warpme/miniarch?tab=readme-ov-file#change-mirror-server-for-udates

Maybe you have issue with data integrity on Eth connection?
May you try download any .zip or tar.bz2 file to your device then unpack it?
(unpacking checks integrity so if there are any data altering on network connection - unpacking will fail)

scoreabasm, 2025-04-21:
As I told you I already tried various mirrors using your recommended procedure after failing of start step 2. To confirm I burned a clean one again to do it at first: same problem.
I don't think I have data integrity on Eth connection, I don't know how to ensure, but strange always when transferring the same files, no? Note that on 136 packages downloaded only few of them are corrupted. I kept the packages cache to do unpack tests with them using 'tar -xvf', I can confirm some are corrupted (ie: gcc-libs-14.2.1+r753+g1cd744a6828f-1-aarch64.pkg.tar.xz) and others are not (ie: zlib-1:1.3.1-2-aarch64.pkg.tar.xz test.tar.xz).  
Still looking for ideas...

warpme, 2025-04-21:
may you give me IP address of host from wich you are downloading i.e. gcc-libs-14.2.1+r753+g1cd744a6828f-1-aarch64.pkg.tar.xz file?

scoreabasm, 2025-04-21:
> may you give me IP address of host from wich you are downloading i.e. gcc-libs-14.2.1+r753+g1cd744a6828f-1-aarch64.pkg.tar.xz file?
Name:	de3.mirror.archlinuxarm.org
Address: 137.226.34.46

warpme, 2025-04-21:
well - I downloaded https://de3.mirror.archlinuxarm.org/aarch64/core/gcc-libs-14.2.1%2Br753%2Bg1cd744a6828f-1-aarch64.pkg.tar.xz
File integrity seems ok.
I can unpack it with xz no problem.
Also local install via pacman -U works ok.

May you try download from:
http://minimyth2.mooo.com/gcc-libs-14.2.1%2Br753%2Bg1cd744a6828f-1-aarch64.pkg.tar.xz
 
then trt local install via packman -U ?

scoreabasm, 2025-04-22:
> well - I downloaded https://de3.mirror.archlinuxarm.org/aarch64/core/gcc-libs-14.2.1%2Br753%2Bg1cd744a6828f-1-aarch64.pkg.tar.xz File integrity seems ok. I can unpack it with xz no problem. Also local install via pacman -U works ok.
> 
> May you try download from: http://minimyth2.mooo.com/gcc-libs-14.2.1%2Br753%2Bg1cd744a6828f-1-aarch64.pkg.tar.xz
> 
> then try local install via packman -U ?

Did a fresh install again + start step 1 then since I had no way to download a file (missing wget before upgrade and upgrade is failing)  downloaded your file from another machine, tested using 'tar -tvf' (ok), then scp to the new machine, and finally tested on this on using 'tar -tvf' wich failed again!! 

I don't understand...

warpme, 2025-04-22:
Well - you say _tested using 'tar -tvf' (ok), then scp to the new machine, and finally tested on this on using 'tar -tvf' wich failed again!!_

I understud:
-download on other machine gives file ok
-this ok file dowloaded via Eth to new machine gives bad file.

So i see 3 hypothesis:
1. Eth download process alters data (so produces error in file)
2. SD read operations on sd card alters data
3. SD write operations on sd card alters data

To judge 1 or 2 you may:
a.put sd card on other machine and write file
b.test file on this other machine with 'tar -tvf'. Repeat a+b ew times to make sure sd card is ok. 
c.reinsert sd to new machine and test with 'tar -tvf' (few times)

If c will be ok - then we know issue in NOT 2
So it can be: hypo1 or hypo3
To test is it 1 or 3:
d. on new machine try copy file with new name and test copied file with 'tar -tvf' (few times: new copy then test)
if d. will be ok - issue is hypo1 
if d. will fail then issue seems hypo3

scoreabasm, 2025-04-25:
> Well - you say _tested using 'tar -tvf' (ok), then scp to the new machine, and finally tested on this on using 'tar -tvf' wich failed again!!_
> 
> I understud: -download on other machine gives file ok -this ok file dowloaded via Eth to new machine gives bad file.
> 
> So i see 3 hypothesis:
> 
> 1. Eth download process alters data (so produces error in file)
> 2. SD read operations on sd card alters data
> 3. SD write operations on sd card alters data
> 
> To judge 1/3 or 2 you may: a.put sd card on other machine and write file b.test file on this other machine with 'tar -tvf'. Repeat a+b ew times to make sure sd card is ok. c.reinsert sd to new machine and test with 'tar -tvf' (few times)
> 
> If c will be ok - then we know issue in NOT 2 So it can be: hypo1 or hypo3 To test is it 1 or 3: d. on new machine try copy file with new name and test copied file with 'tar -tvf' (few times: new copy then test) if d. will be ok - issue is hypo1 if d. will fail then issue seems hypo3
> 
> BTW: what is: -new machine vendor, type -other machine vendor, type -sd card vendor -networking setup

Sorry for the delay, I had to travel and missed time to test. 

So it's neither 1, 2 or 3!!!
Did the following:
- new SD card, burned with your image
- mounted on another machine
- offending files (glibc, gcc-lib, nurses, ....) all downloaded from de mirror, copied to ROOT card partition
- all files tested: OK
- eject the card, mount in new machine and boot the card
- tar -tvf failed!!! 
- eject the card, mount on another machine
- all files tested again (tar -tvf): OK!!!

So not a Eth download problem, not a card read/write problem... look more like a driver/os problem.

Machine vendor: Aliexpress/Electree Store, type:  Orange Pi Zero 3 4GB 
SD Card vendor: Aliexpress/Storage Direct Global Store, type: SanDisk Ultra 32 GB
Networking: Eth0 -> Ethernet cable -> SFR Box (french ISP Fiber offering)

warpme, 2025-04-25:
Oh well...
Lets look for second on yours results:

_new SD card, burned with your image
mounted on another machine
offending files (glibc, gcc-lib, nurses, ....) all downloaded from de mirror, copied to ROOT card partition
all files tested: OK_

this means sd card is ok

_eject the card, mount in new machine and boot the card
tar -tvf failed!!!_

this means for me option2 is suspicious
(SD read operations on sd card alters data)
but as whole system boots ok  - honestly i don't believe option2 it is real hypothesis
 
now this become interesting!

on this zero3 sbc: can you: pack as big as possible bin file then test it with 'tar -tvf' (few times: new copy then test)
this will test r/w ops. on sd card on target

scoreabasm, 2025-04-27:
> Oh well... Lets look for second on yours results:
>
> _new SD card, burned with your image mounted on another machine offending files (glibc, gcc-lib, nurses, ....) all downloaded from de mirror, [...]

Did test writes&reads large files multiple times (tar -cf /usr/bin, tar -cf /usr -> disk full, ... , then tar -tvf... ,  with no problem at all... then start script failed as before at step2: same problem (gcc error... ). Strange no?

warpme, 2025-06-01:
Hi,
I just give test run of current miniarch on mine zero3.
Full install of EOS with Gnome went without issue.
So as it:
1. works for me
2. i have no any other reports about .pkg corruptions on zero3
i want to close this ticket with status "can't reproduce"

scoreabasm, 2025-06-01:
> Hi, I just give test run of current miniarch on mine zero3. Full install of EOS with Gnome went without issue. So as it:
> 
> 1. works for me
> 2. i have no any other reports about .pkg corruptions on zero3
>    i want to close this ticket with status "can't reproduce"