warpme/miniarch #93: Can't connect to wifi network [tanix tx6s]

elpatax, 2025-07-01:
Hello and thanks for this great project,

I installed MiniArch-20240715-6.15.0-board-h616.tanix_tx6s-SD-Image.img on my Tanix Tx6s.

After logging in to xfce I can find my wifi network but I can't connect to it. I've tried the GUI and also nmcli, but none of them work. It just keeps asking for a password.
nmcli returns:

> Error: Connection activation failed: Secrets were required, but not provided.

Here is the boot log:
https://gist.github.com/elpatax/77db873232f0f57bcd8efa8c4133fa66

And also the dmesg output:
https://gist.github.com/elpatax/ee00b0abf12ef563aebd7f3712394ce1

I'd appreciate any help. :)

warpme, 2025-07-05:
Ok - lets see what works at cmd.line...
May you:
1. install iwd (`pacman -S iwd`)
2. try to follow https://github.com/warpme/minimyth2/wiki/Wi-Fi-support-&-configuration#troubleshooting
and provide here results?

elpatax, 2025-07-06:
Thanks for the instructions.

The connection works with iwctl > station wlan0 connect.
I had to create the file /etc/iwd/main.conf (https://wiki.archlinux.org/title/Iwd).

nmcli and the taskbar in xfce show the wifi as disconnected but I can browse the web, so this issue can be closed.

closed 2025-07-06 by elpatax