warpme/miniarch #97: After Upgraded to Kernel-6.16.0 Chromium_v138 Loss VPU Hardware Acceleration

JFLim1, 2025-08-01:
Hi @warpme,

After upgrade to latest kernel-6.16.0, chromium_v138 loss vpu hardware acceleration.  Upgraded from kernel-6.15.4 using:

`sudo start 5`

All 10 reboot/restart chromium_v138 does not have vpu hardware acceleration.  Prior to the upgrade on kernel-6.15.4, chromium_v138 have vpu hardware acceleration at least 50% of the time.

warpme, 2025-08-01:
What HW are you using?
Also may you provide dmesg output?

JFLim1, 2025-08-01:
Device: Orange Pi 5 Plus

```
[alarm@opi5plus ~]$ dmesg -H
[Aug 1 16:42] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[  +0.000000] Linux version 6.16.0 (piotro@minimyth2-aarch64) (aarch64-minimyth-linux-gnu-gcc (GCC) 15.1.0, GNU ld (GNU Binutils) 2.44) #1 SMP PREEMPT Mon Jul 28 15:03:30 CEST 2025
[  +0.000000] Machine model: Xunlong Orange Pi 5 Plus
[  +0.000000] earlycon: uart8250 at MMIO32 0x00000000feb50000 (options '')
[  +0.000000] printk: legacy bootconsole [uart8250] enabled
[  +0.000000] efi: UEFI not found.
[  +0.000000] OF: reserved mem: 0x000000000010f000..0x000000000010f0ff (0 KiB) nomap non-reusable shmem@10f000
[  +0.000000] Zone ranges:
[  +0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[  +0.000000]   DMA32    empty
[  +0.000000]   Normal   [mem 0x0000000100000000-0x00000004ffffffff]
[  +0.000000] Movable zone start for each node                                                                                                                                               
[  +0.000000] Early memory node ranges                                                                                                                                                       
[  +0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]                                                                                                                        
[  +0.000000]   node   0: [mem 0x0000000100000000-0x00000003fbffffff]                                                                                                                        
[  +0.000000]   node   0: [mem 0x00000003fc500000-0x00000003ffefffff]                                                                                                                        
[  +0.000000]   node   0: [mem 0x00000004f0000000-0x00000004ffffffff]                                                                                                                        
[  +0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000004ffffffff]                                                                                                               
[  +0.000000] On node 0, zone DMA: 512 pages in unavailable ranges                                                                                                                           
[  +0.000000] On node 0, zone Normal: 1280 pages in unavailable ranges                                                                                                                       
[  +0.000000] On node 0, zone Normal: 256 pages in unavailable ranges                                                                                                                        
[  +0.000000] cma: Reserved 64 MiB at 0x00000000e8800000                                                                                                                                     
[  +0.000000] psci: probing for conduit method from DT.                                                                                                                                      
[  +0.000000] psci: PSCIv1.1 detected in firmware.                                                                                                                                           
[  +0.000000] psci: Using standard PSCI v0.2 function IDs                                                                                                                                    
[  +0.000000] psci: MIGRATE_INFO_TYPE not supported.                                                                                                                                         
[  +0.000000] psci: SMC Calling Convention v1.2                                                                                                                                              
[  +0.000000] No bootconfig data provided, so skipping bootconfig                                                                                                                            
[  +0.000000] percpu: Embedded 33 pages/cpu s95064 r8192 d31912 u135168                                                                                                                      
[  +0.000000] pcpu-alloc: s95064 r8192 d31912 u135168 alloc=33*4096                                                                                                                          
[  +0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7                                                                                                                    
[  +0.000000] Detected VIPT I-cache on CPU0                                                                                                                                                  
[  +0.000000] CPU features: detected: GIC system register CPU interface                                                                                                                      
[  +0.000000] CPU features: detected: Virtualization Host Extensions                                                                                                                         
[  +0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009                                                                                                 
[... cut ...]
[  +0.003517] rockchip-drm display-subsystem: swiotlb buffer is full (sz: 1073152 bytes), total 32768 (slots), used 73 (slots)
[  +0.700609] rockchip-drm display-subsystem: swiotlb buffer is full (sz: 1073152 bytes), total 32768 (slots), used 13 (slots)
[  +0.013874] rockchip-drm display-subsystem: swiotlb buffer is full (sz: 962560 bytes), total 32768 (slots), used 5 (slots)
[  +0.016644] rockchip-drm display-subsystem: swiotlb buffer is full (sz: 851968 bytes), total 32768 (slots), used 13 (slots)
(END)
```

warpme, 2025-08-16:
Hi,
I suspect your issues might be related to interferences between on-board spi stored vendor bootloader with sd card provided mainline bootloader. (e.g. I have such issue on 2 rk3588 boards in my lab).
I uploaded new images with updated bootloader where such interference seems to be fixed.
you may try and see how it goes for you.
(to test this you need boot device from sd card and do fresh install on emmc - if you are using emmc...)

JFLim1, 2025-08-16:
> I uploaded new images with updated bootloader where such interference seems to be fixed.
> you may try and see how it goes for you.

Thank you.  Will update to your new kernel-6.16.1 will do the trick or do I have to reflash the SD Card with your new image v14.4.0?

Is there a way to flash just the updated bootloader on the current exiting SD Card?  Instructions/commands and where to download the updated bootloader?

warpme, 2025-08-16:
well - actual bootloader files are stored on sd card boot partition in` /bootloader` dir.
So you may update bootloader on existing sd bard by:
1. download 14.4.0 sd card image
2. mount image and copy bootloader files from `/bootloader` dir to your existing card dir.
3. start terminal and type "`install-on-emmc bootloader`" this should update bootloader on existing sdcard with new bootloader
4. reboot and see uart out: you should see 2025.10-rc1 uboot ver.

JFLim1, 2025-08-16:
Am basically a greenhorn on linux.

Thanks for the guide to copy the updated bootloader from the new MiniArch_v14.4.0.  Downloaded MiniArch_v14.4.0 onto NVme SSD (/run/media/alarm/writable/home/jfl/Downloads/MiniArch/MiniArch-20240715-6.16.1-board-rk3588.orangepi_5_plus-SD-Image.img.xz)

Tried to mount the newly downloaded MiniArch image with the command below but unsuccessful.

```
[alarm@opi5plus ~]$ sudo mount -o loop /run/media/alarm/writable/home/jfl/Downloads/MiniArch/MiniArch-20240715-6.16.1-board-rk3588.orangepi_5_plus-SD-Image.img.xz  /mnt/
[sudo] password for alarm: 
mount: /mnt: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
[alarm@opi5plus ~]$ 
```
Wasn't using the correct command to mount the new MiniArch_v14.4.0 image.  Don't mind sharing the correct command to mount MiniArch_v14.4.0 image and copy the new bootloader to the existing MiniArch image.

warpme, 2025-08-17:
As we are not sure will new bootloader fix your gpu issues - maybe good will be to do first just quick test with new image flashed to blank sdcard and see how it goes?
If it will go ok - then we you can update bootloader on existing card?

closed 2025-08-19 by JFLim1

JFLim1, 2025-08-20:
Hi @warpme,

Out of curiosity transplanted kernel-6.16.1-koma-3pole.tar.gz compiled by Tripole (https://mega.nz/folder/GJNjjSAY#Zru4DqCa6hyYnnkGuw02Rw/file/WV9FHDyT) on Manjaro-opi5plus image, this kernel-6.16.1-koma-3pole support vpu hardware acceleration on Chromium, clapper and mpv-full-git (with v4l2request) and also seem to resolve the KDE Plasma mouse visibility issue.

Kernel-6.16.1-koma-3pole does not support hevc, 4k video on clapper or mpv-full-git it only has hantro_vpu support where your kernel-6.16.1.

warpme, 2025-09-22:
I added https://github.com/armbian/build/commit/4f269906335a2c7e7063c2f16a2689cd08e40375 to just published 6.16.8.
may you update and see how it goes?

JFLim1, 2025-09-24:
Not sure whether this is useful info.  Fedora-42_Workstation with Fedora kernel-6.16.7-200.fc42. Chromium always has vpu hardware acceleration. 

```
jfl@anonymous:~$ echo "v4l2 video devices"
v4l2_devices=`ls -1 /dev/video*`
for d in ${v4l2_devices}; do
  echo "--------------------------------------------------------------------------------"
  echo "Device: ${d}:"
  v4l2-ctl -d ${d} -D --list-formats-ext --list-formats-out-ext
  echo "--------------------------------------------------------------------------------"
  echo " "
done
v4l2 video devices
--------------------------------------------------------------------------------
Device: /dev/video0:
Driver Info:
	Driver name      : rockchip-rga
	Card type        : rockchip-rga
	Bus info         : platform:rga
	Driver version   : 6.16.7
	Capabilities     : 0x84204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'BA24' (32-bit ARGB 8-8-8-8)
	[1]: 'AR24' (32-bit BGRA 8-8-8-8)
	[2]: 'XR24' (32-bit BGRX 8-8-8-8)
	[3]: 'RGB3' (24-bit RGB 8-8-8)
	[4]: 'BGR3' (24-bit BGR 8-8-8)
	[5]: 'AR12' (16-bit ARGB 4-4-4-4)
	[6]: 'AR15' (16-bit ARGB 1-5-5-5)
	[7]: 'RGBP' (16-bit RGB 5-6-5)
	[8]: 'NV21' (Y/VU 4:2:0)
	[9]: 'NV61' (Y/VU 4:2:2)
	[10]: 'NV12' (Y/UV 4:2:0)
	[11]: 'NM12' (Y/UV 4:2:0 (N-C))
	[12]: 'NV16' (Y/UV 4:2:2)
	[13]: 'YU12' (Planar YUV 4:2:0)
	[14]: '422P' (Planar YUV 4:2:2)
	[15]: 'YV12' (Planar YVU 4:2:0)
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output Multiplanar

	[0]: 'BA24' (32-bit ARGB 8-8-8-8)
	[1]: 'AR24' (32-bit BGRA 8-8-8-8)
	[2]: 'XR24' (32-bit BGRX 8-8-8-8)
	[3]: 'RGB3' (24-bit RGB 8-8-8)
	[4]: 'BGR3' (24-bit BGR 8-8-8)
	[5]: 'AR12' (16-bit ARGB 4-4-4-4)
	[6]: 'AR15' (16-bit ARGB 1-5-5-5)
	[7]: 'RGBP' (16-bit RGB 5-6-5)
	[8]: 'NV21' (Y/VU 4:2:0)
	[9]: 'NV61' (Y/VU 4:2:2)
	[10]: 'NV12' (Y/UV 4:2:0)
	[11]: 'NM12' (Y/UV 4:2:0 (N-C))
	[12]: 'NV16' (Y/UV 4:2:2)
	[13]: 'YU12' (Planar YUV 4:2:0)
	[14]: '422P' (Planar YUV 4:2:2)
	[15]: 'YV12' (Planar YVU 4:2:0)
--------------------------------------------------------------------------------
 
--------------------------------------------------------------------------------
[... 116 lines cut ...]
	Hardware revision: 0x00000000 (0)
	Driver version   : 6.16.7
Interface Info:
	ID               : 0x0300000c
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : rockchip,rk3588-av1-vpu-dec-source
	Function         : V4L2 I/O
	Pad 0x01000002   : 0: Source
	  Link 0x02000008: to remote pad 0x1000004 of entity 'rockchip,rk3588-av1-vpu-dec-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'VT12' (Y/UV 4:2:0 (4x4 Linear))
	[1]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output Multiplanar

	[0]: 'AV1F' (AV1 Frame, compressed)
		Size: Stepwise 64x64 - 4096x2304 with step 16/16
--------------------------------------------------------------------------------
 
jfl@anonymous:~$ 
```

warpme, 2025-09-25:
I really appreciate you help with solving this issue!
I see all cases where chrominium decode works are cases where h264 decoder is hantro.
In miniarch kernel i temp. disabled hantro (which effectively forces userspace to use rkvdec2 h264 decoder) and this seems to be problematic for chrominuim (while works ok for other players).
it is still not clear for me: do we have any 6.16 kernel offering working in chrominuim rkvdec2 h264 decoder? 
Is such kernel is armbian kernel-6.16.7-edge-rockchip64?

JFLim1, 2025-09-26:
Hi @warpme,

Collabora kernel does have "rkvdec2" and hantro-vpu-rk3568  and on every boot up vpu hw acceleration is available on Chromium.

```
[jfl@jfl-opi5plus ~]$ uname -a
Linux jfl-opi5plus 6.16.0-1-aarch64-rk3588-collabora-git-g1e63f2aa9305 #1 SMP PREEMPT Thu Sep  4 21:47:26 UTC 2025 aarch64 GNU/Linux
[jfl@jfl-opi5plus ~]$ 

[jfl@jfl-opi5plus ~]$ echo "v4l2 video devices"
v4l2_devices=`ls -1 /dev/video*`
for d in ${v4l2_devices}; do
  echo "--------------------------------------------------------------------------------"
  echo "Device: ${d}:"
  v4l2-ctl -d ${d} -D --list-formats-ext --list-formats-out-ext
  echo "--------------------------------------------------------------------------------"
  echo " "
done
v4l2 video devices
--------------------------------------------------------------------------------
Device: /dev/video0:
Driver Info:
	Driver name      : rockchip-rga
	Card type        : rockchip-rga
	Bus info         : platform:rga
	Driver version   : 6.16.0
	Capabilities     : 0x84204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'BA24' (32-bit ARGB 8-8-8-8)
	[1]: 'AR24' (32-bit BGRA 8-8-8-8)
	[2]: 'XR24' (32-bit BGRX 8-8-8-8)
	[3]: 'RGB3' (24-bit RGB 8-8-8)
	[4]: 'BGR3' (24-bit BGR 8-8-8)
	[5]: 'AR12' (16-bit ARGB 4-4-4-4)
	[6]: 'AR15' (16-bit ARGB 1-5-5-5)
	[7]: 'RGBP' (16-bit RGB 5-6-5)
	[8]: 'NV21' (Y/VU 4:2:0)
	[9]: 'NV61' (Y/VU 4:2:2)
	[10]: 'NV12' (Y/UV 4:2:0)
	[11]: 'NM12' (Y/UV 4:2:0 (N-C))
	[12]: 'NV16' (Y/UV 4:2:2)
	[13]: 'YU12' (Planar YUV 4:2:0)
	[14]: '422P' (Planar YUV 4:2:2)
	[15]: 'YV12' (Planar YVU 4:2:0)
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output Multiplanar

	[0]: 'BA24' (32-bit ARGB 8-8-8-8)
	[1]: 'AR24' (32-bit BGRA 8-8-8-8)
	[2]: 'XR24' (32-bit BGRX 8-8-8-8)
	[3]: 'RGB3' (24-bit RGB 8-8-8)
	[4]: 'BGR3' (24-bit BGR 8-8-8)
	[5]: 'AR12' (16-bit ARGB 4-4-4-4)
	[6]: 'AR15' (16-bit ARGB 1-5-5-5)
	[7]: 'RGBP' (16-bit RGB 5-6-5)
	[8]: 'NV21' (Y/VU 4:2:0)
	[9]: 'NV61' (Y/VU 4:2:2)
	[10]: 'NV12' (Y/UV 4:2:0)
	[11]: 'NM12' (Y/UV 4:2:0 (N-C))
	[12]: 'NV16' (Y/UV 4:2:2)
[... 256 lines cut ...]
	Driver version   : 6.16.0
Interface Info:
	ID               : 0x0300000c
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : rockchip,rk3568-vpu-dec-source
	Function         : V4L2 I/O
	Pad 0x01000002   : 0: Source
	  Link 0x02000008: to remote pad 0x1000004 of entity 'rockchip,rk3568-vpu-dec-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output Multiplanar

	[0]: 'S264' (H.264 Parsed Slice Data, compressed)
		Size: Stepwise 48x48 - 1920x1088 with step 16/16
	[1]: 'MG2S' (MPEG-2 Parsed Slice Data, compressed)
		Size: Stepwise 48x48 - 1920x1088 with step 16/16
	[2]: 'VP8F' (VP8 Frame, compressed)
		Size: Stepwise 48x48 - 3840x2160 with step 16/16
--------------------------------------------------------------------------------
```

JFLim1, 2025-09-30:
Hi @warpme,

Good news is that with kernel-6.17-1 there is a chance Chromium have vpu hardware acceleration when boot-up with:

```
Device: /dev/video0:
Driver Info:
        Driver name      : hantro-vpu
        Card type        : rockchip,rk3568-vpu-dec
```

The bad news is if on boot up:

```
Device: /dev/video0:
Driver Info:
        Driver name      : rkvdec
        Card type        : rkvdec
```

or /dev/video9: is other than hantro-vpu -- rockchip,rk3568-vpu-dec, Chromium will NOT have vpu hardware acceleration.  This is similar/same behavior as on kernel-6.15.4.

Am not sure whether this play a part for Chromium to be able to use both hantro-vpu and rkvdec to decode a both armbian kernel-6.16.8-edge-rockchip64 and collabora kernel-6.16.0-git both have "rockchip-rga" driver..

```
Device: /dev/video0:
Driver Info:
	Driver name      : rockchip-rga
	Card type        : rockchip-rga
	Bus info         : platform:rga
```

On armbian kernel-6.16.8-edge-rockchip64 and Collabora kernel-6.16.0-git, Chromium have vpu acceleration every boot up regardless whether /dev/video0: is hantro-vpu or rkvdec or rockchip-rga.

```
Device: /dev/video0:
Driver Info:
	Driver name      : rockchip-rga
	Card type        : rockchip-rga
	Bus info         : platform:rga
```

Not sure whether having "rockchip-rga" will make a differnece.

warpme, 2025-10-01:
may to pls save below as `99-chromium-video-decoders.rules` and drop it to `/etc/udev/rules.d` 

```
SUBSYSTEM=="video4linux", ATTR{name}=="rockchip,rk3568-vpu-dec", SYMLINK+="video-dec%n"
SUBSYSTEM=="video4linux", ATTR{name}=="rockchip,rk3588-av1-vpu-dec", SYMLINK+="video-dec%n"
SUBSYSTEM=="media", ATTR{model}=="hantro-vpu", SYMLINK+="media-dec%n"
```

then reboot and test?

JFLim1, 2025-10-02:
Thanks.  Will edit accordingly.

Before Edit: "90-chromium-video.rules" 
```
# This is qcom venus v4l2 stateful decoder, tested on sm8250
SUBSYSTEM=="video4linux", ATTR{name}=="qcom-venus-decoder", SYMLINK+="video-dec%n"

# This is qcom venus v4l2 stateful encoder, not tested yet, but chromium did query /dev/video-enc* devives for v4l2 encoding.
SUBSYSTEM=="video4linux", ATTR{name}=="qcom-venus-encoder", SYMLINK+="video-enc%n"

# These rules are for rkvdec v4l2 stateless decoder found on rk3399
SUBSYSTEM=="video4linux", ATTR{name}=="rkvdec", SYMLINK+="video-dec%n"
SUBSYSTEM=="media", ATTR{model}=="rkvdec", SYMLINK+="media-dec%n"

# These rules are for rkvdec2 v4l2 stateless decoder found on rk35xx
SUBSYSTEM=="video4linux", ATTR{name}=="rkvdec2", SYMLINK+="video-dec%n"
SUBSYSTEM=="media", ATTR{model}=="rkvdec2", SYMLINK+="media-dec%n"

# These rules are for hantro v4l2 stateless decoders found on rk3328, rk3399, rk3568 and rk3588
SUBSYSTEM=="video4linux", ATTR{name}=="rockchip,rk3328-vpu-dec", SYMLINK+="video-dec%n"
SUBSYSTEM=="video4linux", ATTR{name}=="rockchip,rk3399-vpu-dec", SYMLINK+="video-dec%n"
SUBSYSTEM=="video4linux", ATTR{name}=="rockchip,rk3568-vpu-dec", SYMLINK+="video-dec%n"
# AV1 decoder on rk3588 is still not well supported, so disable it now
#SUBSYSTEM=="video4linux", ATTR{name}=="rockchip,rk3588-av1-vpu-dec", SYMLINK+="video-dec%n"
SUBSYSTEM=="media", ATTR{model}=="hantro-vpu", SYMLINK+="media-dec%n"

# These rules are for Amlogic meson vdec stateful decoder found on S922X
SUBSYSTEM=="video4linux", ATTR{name}=="meson-video-decoder", SYMLINK+="video-dec%n"
#SUBSYSTEM=="video4linux", ATTR{name}=="meson_vdec", SYMLINK+="video-dec%n"
#SUBSYSTEM=="video4linux", ATTR{name}=="vdec", SYMLINK+="video-dec%n"
#SUBSYSTEM=="media", ATTR{model}=="vdec", SYMLINK+="media-dec%n"
SUBSYSTEM=="media", ATTR{model}=="meson_vdec", SYMLINK+="media-dec%n"
```

After Edit:  "90-chromium-video.rules" the changes:

```
# These rules are for hantro v4l2 stateless decoders found on rk3328, rk3399, rk3568 and rk3588
SUBSYSTEM=="video4linux", ATTR{name}=="rockchip,rk3568-vpu-dec", SYMLINK+="video-dec%n"
SUBSYSTEM=="video4linux", ATTR{name}=="rockchip,rk3588-av1-vpu-dec", SYMLINK+="video-dec%n"
SUBSYSTEM=="media", ATTR{model}=="hantro-vpu", SYMLINK+="media-dec%n"
```

Will revert after testing to see whether it will result in Chromium having vpu hw acceleration on every boot-up.

JFLim1, 2025-10-03:
Now I have startup or rebooted in total 12 times.  Results 
7 times /dev/video0: rk3568-vpu-dec --Chromium has vpu hardware acceleration
5 times /dev/video0: rkvdec -- Chromium NO vpu hardware acceleration

The assignment of which video driver to /dev/video0 /dev/video1, etc doesn't seem to be related to "90-chromium-video.rules".  Assignment of video decoder (Device: /dev/video-dec1:) is related to  "90-chromium-video.rules".  But for yur kernel-6.17-1 only when /dev/video: rk3568-vpu-dec is set/assigned then Chromium vpu hardware acceleration is available.

Where as on kernel-6.16.y-edge-rockchip64 so far whether /dev/video0: rockchip-rga, rk3328-vpu-dec or rkvdec, vpu hardware acceleration is available on Chromium on every boot up.  

On collabora-git kernel similar to edge-rockchip64.

```
[alarm@opi5plus ~]$ echo "v4l2 video devices"
v4l2_devices=`ls -1 /dev/video*`
for d in ${v4l2_devices}; do
  echo "--------------------------------------------------------------------------------"
  echo "Device: ${d}:"
  v4l2-ctl -d ${d} -D --list-formats-ext --list-formats-out-ext
  echo "--------------------------------------------------------------------------------"
  echo " "
done
v4l2 video devices
--------------------------------------------------------------------------------
Device: /dev/video0:
Driver Info:
        Driver name      : rkvdec
        Card type        : rkvdec
        Bus info         : platform:rkvdec
        Driver version   : 6.17.0
        Capabilities     : 0x84204000
                Video Memory-to-Memory Multiplanar
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04204000
                Video Memory-to-Memory Multiplanar
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : rkvdec
        Model            : rkvdec
        Serial           : 
        Bus info         : platform:rkvdec
        Media version    : 6.17.0
        Hardware revision: 0x00000000 (0)
        Driver version   : 6.17.0
Interface Info:
        ID               : 0x0300000c
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : rkvdec-source
        Function         : V4L2 I/O
        Pad 0x01000002   : 0: Source
          Link 0x02000008: to remote pad 0x1000004 of entity 'rkvdec-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture Multiplanar

        [0]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'S264' (H.264 Parsed Slice Data, compressed)
                Size: Continuous 1x1 - 65520x65520
        [1]: 'S265' (HEVC Parsed Slice Data, compressed)
                Size: Continuous 1x1 - 65472x65472
--------------------------------------------------------------------------------
 
--------------------------------------------------------------------------------
Device: /dev/video1:
Driver Info:
[... 207 lines cut ...]
        Hardware revision: 0x00000000 (0)
        Driver version   : 6.17.0
Interface Info:
        ID               : 0x0300000c
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : rockchip,rk3588-av1-vpu-dec-source
        Function         : V4L2 I/O
        Pad 0x01000002   : 0: Source
          Link 0x02000008: to remote pad 0x1000004 of entity 'rockchip,rk3588-av1-vpu-dec-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture Multiplanar

        [0]: 'VT12' (Y/UV 4:2:0 (4x4 Linear))
        [1]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'AV1F' (AV1 Frame, compressed)
                Size: Stepwise 64x64 - 4096x2304 with step 16/16
--------------------------------------------------------------------------------
 
[alarm@opi5plus ~]$
```

warpme, 2025-10-17:
ah good!
Pls play/use few days - to be sure things are really ok....

JFLim1, 2025-10-18:
Will do and update.

JFLim1, 2025-12-14:
Hi @warpme,
Just a feedback.  Upgraded to your latest kernel-6.18.1 on MiniArch-EOS-Gnome.  MiniArch-EOS-Gnome runs stable so far.

One  interesting observation with kernel-6.18.0/6.18.1 (can't remember whether 6.17.6-1 behave the same) if /dev/video1: rkvdec and /dev/video2: hantaro-vpu rk3568-vpu-dec (or anytime rkvdec is loaded ahead of hantro-vpu rk3568-vpu-dec)  Chromium will be able to play upt to h264 4k/60 video.

If example /dev/video1: hantro-vpu rk3568-vpu-dec and /dev/video2: rkvdec (or anytime hantro-vpu rk3568-vpu-dec is loaded ahead of rkvdec) Chromium will can only support up to h264 1080p/60.

In this example /dev/video1: hantro-vpu  rk3568-vpu-dec and /dev/video5: rkvdec.  Chromium can only support up to 1080p/60 h264
[image]

JFLim1, 2025-12-14:
For your info (may be you might have already noticed it) there is a kernel patch to allow stable 1080p/60 av1 video to play/stream on Clapper (and on a patched Chromium did have a patched Chromium so could not test) stable without crashing for example Big Buck Bunny 1080/60 av1 https://lafibre.info/videos/test/201411_blender_big_buck_bunny_60fps_1080p_av1.mp4.

With the patch https://lafibre.info/videos/test/201411_blender_big_buck_bunny_60fps_1080p_av1.mp4 will stop playing after 44 second on Clapper.

The patch https://github.com/armbian/build/commit/3dbd4e8e092aa05cb9fcb4db54550df1e0998737

warpme, 2025-12-15:
> Without the patch https://lafibre.info/videos/test/201411_blender_big_buck_bunny_60fps_1080p_av1.mp4 will stop playing after 44 second on Clapper. With the patch Clapper will stream this 1080p/60 av1 video all the way to end (10 mins+),
> 
> The patch [armbian/build@3dbd4e8](https://github.com/armbian/build/commit/3dbd4e8e092aa05cb9fcb4db54550df1e0998737)

Ah - many thx for your constant testing and hints!
For sure i will include this patch in next kernel release :-)