warpme/miniarch #98: After Update To Kernel-6.16.0 Mouse Cursor/Pointer No Longer Visible On KDE Plasma-Wayland

JFLim1, 2025-08-01:
Device: Orange Pi 5 Plus

After Updating to Kernel-6.16.0 mouse cursor/pointer no longer visible on KDE Plasma-Wayland Session.  Mouse cursor/pointer is visible on KDE Plasma-X11 Session.  Same issue for LXQT-Wayland-Kwin session.  

Mouse pointer is visible on LXQT-Wayland-labwc and LXQT-X11 session.

A work around for mouse cursor/pointer to be visible on KDE Plasma-Wayland session is to include:

`export KWIN_FORCE_SW_CURSOR=1`

into /etc/environment

warpme, 2025-08-16:
Hi,
I suspect your issues might be related to interferences between on-board spi stored vendor bootloader with sd card provided mainline bootloader. (e.g. I have such issue on 2 rk3588 boards in my lab).
I uploaded new images with updated bootloader where such interference seems to be fixed.
you may try and see how it goes for you.
(to test this you need boot device from sd card and do fresh install on emmc - if you are using emmc...)

warpme, 2025-08-20:
May you try: http://minimyth2.mooo.com/linux-aarch64-6.16.1-2-any.pkg.tar.gz ?

JFLim1, 2025-08-22:
> May you try: http://minimyth2.mooo.com/linux-aarch64-6.16.1-2-any.pkg.tar.gz ?

It requested for User and Password.  I don't have user account on minimyth2.moo.com.  github account did not work.

warpme, 2025-08-22:
Ah sorry. May you try now?

JFLim1, 2025-08-22:
Thank you.  Downloaded and installed linux-aarch64 6.16.1-2 successfully (but still using the old kernel-headers-6.16.1-1).  

```
[alarm@opi5plus ~]$ uname -a
Linux opi5plus 6.16.1 #1 SMP PREEMPT Wed Aug 20 17:26:27 CEST 2025 aarch64 GNU/Linux

[alarm@opi5plus ~]$ pacman -Qs linux-aarch64
local/linux-aarch64 6.16.1-2
    The Linux Kernel and modules - AArch64 multi-platform
local/linux-aarch64-api-headers 6.16.1-1
    
local/linux-aarch64-headers 6.16.1-1
    Header files and scripts for building modules for linux kernel - AArch64
    multi-platform
[alarm@opi5plus ~]$ 
```

```
/usr/share/sddm/scripts/wayland-session 
# Restore user shell setting that may have been clobbered by setting environment
export SHELL=$SDDM_USER_SHELL
#export KWIN_FORCE_SW_CURSOR=1

exec $@
```

Thnak you.  Good news is that with the new kernel-6.16.1-2 the mouse cursor/pointer is visible in KDE Plasma Desktop without having to force software cursor.

But still no vpu hardware acceleration on Chromium (4 reboots/power-up).

JFLim1, 2025-08-22:
Hi @warpme,

On MiniArch-EOS-KDE with your kernel 6.15.4 (after 4 reboot with no chromium vpu hw acceleration available). Shutdown and power-off and power-on again to restart.  This time chromium have vpu hardware acceleration and the v4l2-request decoder sequence is different from the no vpu hw acceleration.

Kernel-6.15.4  where chromium has vpu hardware acceleration

```
[alarm@opi5plus ~]$ echo "v4l2 video devices"
v4l2_devices=`ls -1 /dev/video*`
for d in ${v4l2_devices}; do
  echo "--------------------------------------------------------------------------------"
  echo "Device: ${d}:"
  v4l2-ctl -d ${d} -D --list-formats-ext --list-formats-out-ext
  echo "--------------------------------------------------------------------------------"
  echo " "
done
v4l2 video devices
--------------------------------------------------------------------------------
Device: /dev/video0:
Driver Info:
        Driver name      : hantro-vpu
        Card type        : rockchip,rk3568-vpu-dec
        Bus info         : platform:fdb50000.video-codec
        Driver version   : 6.15.4
        Capabilities     : 0x84204000
                Video Memory-to-Memory Multiplanar
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04204000
                Video Memory-to-Memory Multiplanar
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : hantro-vpu
        Model            : hantro-vpu
        Serial           : 
        Bus info         : platform:fdb50000.video-codec
        Media version    : 6.15.4
        Hardware revision: 0x00000000 (0)
        Driver version   : 6.15.4
Interface Info:
        ID               : 0x0300000c
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : rockchip,rk3568-vpu-dec-source
        Function         : V4L2 I/O
        Pad 0x01000002   : 0: Source
          Link 0x02000008: to remote pad 0x1000004 of entity 'rockchip,rk3568-vpu-dec-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture Multiplanar

        [0]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'S264' (H.264 Parsed Slice Data, compressed)
                Size: Stepwise 48x48 - 1920x1088 with step 16/16
        [1]: 'MG2S' (MPEG-2 Parsed Slice Data, compressed)
                Size: Stepwise 48x48 - 1920x1088 with step 16/16
        [2]: 'VP8F' (VP8 Frame, compressed)
                Size: Stepwise 48x48 - 3840x2160 with step 16/16
--------------------------------------------------------------------------------
 
--------------------------------------------------------------------------------
Device: /dev/video1:
Driver Info:
        Driver name      : hantro-vpu
        Card type        : rockchip,rk3588-vepu121-enc
[... 156 lines cut ...]
        Driver version   : 6.15.4
Interface Info:
        ID               : 0x0300000c
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : rkvdec-source
        Function         : V4L2 I/O
        Pad 0x01000002   : 0: Source
          Link 0x02000008: to remote pad 0x1000004 of entity 'rkvdec-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture Multiplanar

        [0]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'S264' (H.264 Parsed Slice Data, compressed)
                Size: Continuous 1x1 - 65520x65520
        [1]: 'S265' (HEVC Parsed Slice Data, compressed)
                Size: Continuous 1x1 - 65472x65472
--------------------------------------------------------------------------------
 
[alarm@opi5plus ~]$ 
```

JFLim1, 2025-08-22:
Manjaro-opi5plus-xfce switch to kernel 6.15.0-1-aarch64-rk3588-collabora-git-g33755850faeb.  This kernel 6.15.0-1-collabora-git does NOT support hevc or 4K h264 video.  This kernel has "rkvdec2" decoder.

```
[jfl@jfl-opi5plus ~]$ uname -a
Linux jfl-opi5plus 6.15.0-1-aarch64-rk3588-collabora-git-g33755850faeb #1 SMP PREEMPT Mon Jul  7 21:47:51 UTC 2025 aarch64 GNU/Linux

[jfl@jfl-opi5plus ~]$ echo "v4l2 video devices"
v4l2_devices=`ls -1 /dev/video*`
for d in ${v4l2_devices}; do
  echo "--------------------------------------------------------------------------------"
  echo "Device: ${d}:"
  v4l2-ctl -d ${d} -D --list-formats-ext --list-formats-out-ext
  echo "--------------------------------------------------------------------------------"
  echo " "
done
v4l2 video devices
--------------------------------------------------------------------------------
Device: /dev/video0:
Driver Info:
	Driver name      : rkvdec2
	Card type        : rkvdec2
	Bus info         : platform:rkvdec2
	Driver version   : 6.15.0
	Capabilities     : 0x84204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : rkvdec2
	Model            : rkvdec2
	Serial           : 
	Bus info         : platform:rkvdec2
	Media version    : 6.15.0
	Hardware revision: 0x00000000 (0)
	Driver version   : 6.15.0
Interface Info:
	ID               : 0x0300000c
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : rkvdec2-source
	Function         : V4L2 I/O
	Pad 0x01000002   : 0: Source
	  Link 0x02000008: to remote pad 0x1000004 of entity 'rkvdec2-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output Multiplanar

	[0]: 'S264' (H.264 Parsed Slice Data, compressed)
		Size: Continuous 1x1 - 65520x65520
--------------------------------------------------------------------------------
 
--------------------------------------------------------------------------------
Device: /dev/video1:
Driver Info:
	Driver name      : hantro-vpu
	Card type        : rockchip,rk3568-vpu-dec
	Bus info         : platform:fdb50000.video-codec
	Driver version   : 6.15.0
	Capabilities     : 0x84204000
		Video Memory-to-Memory Multiplanar
		Streaming
[... 107 lines cut ...]
	Hardware revision: 0x00000000 (0)
	Driver version   : 6.15.0
Interface Info:
	ID               : 0x0300000c
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : rockchip,rk3588-av1-vpu-dec-source
	Function         : V4L2 I/O
	Pad 0x01000002   : 0: Source
	  Link 0x02000008: to remote pad 0x1000004 of entity 'rockchip,rk3588-av1-vpu-dec-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'VT12' (Y/UV 4:2:0 (4x4 Linear))
	[1]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output Multiplanar

	[0]: 'AV1F' (AV1 Frame, compressed)
		Size: Stepwise 64x64 - 3840x2160 with step 16/16
--------------------------------------------------------------------------------
 
[jfl@jfl-opi5plus ~]$
```

JFLim1, 2025-08-25:
Hi @warpme,

After three days of testing, in summary with your kernel-6.15.4 or Manjaro kernel linux-6.15.8 

Chromium will NOT have vpu acceleration whenever:
/dev/video0: rkvdec
or
/dev/video0: hantro-vpu, rockchip,rk3588-vepu121-enc

Chromium will have vpu hardware acceleration whenever:

/dev/video0: hantro-vpu, rockchip,rk3568-vpu-dec

Your new kernel-6.16.2 does NOT seem to have "hantro-vpu, rockchip,rk3568-vpu-dec" driver/firmware as such, chromium does NOT have vpu hardware acceleration.

Hopefully these info are helpful.

JFLim1, 2025-08-28:
Hi, Debian-Trixie with kernel 6.16.3-edge-rockchip64 (armbian kernel).  Interestingly this kernel have additional settings: in addition to the usual /dev/video0 to /dev/video4

Chromiur has vpu hardware acceleration on every boot up.

```
Device: /dev/video-dec1:
Driver Info:
	Driver name      : hantro-vpu
	Card type        : rockchip,rk3328-vpu-dec
	Bus info         : platform:fdb50000.video-codec
	Driver version   : 6.16.3
	Capabilities     : 0x84204000
		Video Memory-to-Memory Multiplanar

Device: /dev/video-dec4:
Driver Info:
	Driver name      : rkvdec2
	Card type        : rkvdec2
	Bus info         : platform:rkvdec2
	Driver version   : 6.16.3
	Capabilities     : 0x84204000
		Video Memory-to-Memory Multiplanar

```

```
jfl@jfl-orangepi5plus:~$ echo "v4l2 video devices"
v4l2_devices=`ls -1 /dev/video*`
for d in ${v4l2_devices}; do
  echo "--------------------------------------------------------------------------------"
  echo "Device: ${d}:"
  v4l2-ctl -d ${d} -D --list-formats-ext --list-formats-out-ext
  echo "--------------------------------------------------------------------------------"
  echo " "
done
v4l2 video devices
--------------------------------------------------------------------------------
Device: /dev/video0:
Driver Info:
	Driver name      : rockchip-rga
	Card type        : rockchip-rga
	Bus info         : platform:rga
	Driver version   : 6.16.3
	Capabilities     : 0x84204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'BA24' (32-bit ARGB 8-8-8-8)
	[1]: 'AR24' (32-bit BGRA 8-8-8-8)
	[2]: 'XR24' (32-bit BGRX 8-8-8-8)
	[3]: 'RGB3' (24-bit RGB 8-8-8)
	[4]: 'BGR3' (24-bit BGR 8-8-8)
	[5]: 'AR12' (16-bit ARGB 4-4-4-4)
	[6]: 'AR15' (16-bit ARGB 1-5-5-5)
	[7]: 'RGBP' (16-bit RGB 5-6-5)
	[8]: 'NV21' (Y/VU 4:2:0)
	[9]: 'NV61' (Y/VU 4:2:2)
	[10]: 'NV12' (Y/UV 4:2:0)
	[11]: 'NM12' (Y/UV 4:2:0 (N-C))
	[12]: 'NV16' (Y/UV 4:2:2)
	[13]: 'YU12' (Planar YUV 4:2:0)
	[14]: '422P' (Planar YUV 4:2:2)
[... 271 lines cut ...]
	Media version    : 6.16.3
	Hardware revision: 0x00000000 (0)
	Driver version   : 6.16.3
Interface Info:
	ID               : 0x0300000c
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : rkvdec2-source
	Function         : V4L2 I/O
	Pad 0x01000002   : 0: Source
	  Link 0x02000008: to remote pad 0x1000004 of entity 'rkvdec2-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output Multiplanar

	[0]: 'S264' (H.264 Parsed Slice Data, compressed)
		Size: Stepwise 16x16 - 65520x65520 with step 16/16
--------------------------------------------------------------------------------
 
jfl@jfl-orangepi5plus:~$
```

warpme, 2025-09-22:
Sorry for long silence!
Issue with sw cursor is kind of trade-off I must handle and is result of drm planes reordering.
reordering is required to get kodi & mythtv video playback ok. (without it - users are getting black screen at playback)
unfortunately this reordering causes hw cursor issue in wayland...
So I decided to go with less worse solution: do reordering + req. to use sw cursor in wayland.
(alternative decision gives hw cursor but breaks kodi/mythtv)
hope you are agree with this decision :-)

JFLim1, 2025-09-22:
For troubleshooting purpose. 

Transplated kernel-6.16.7-edge-rockchip64 to MiniArch-EOS-KDE image but using dtb from your kernel-6.16.8-1 as if dtb from kernel-6.16.7-edge-rockchip64 is used it would not boot up successfully to KDE Plasma Desktop.

Chromium always have vpu acceleration with kernel-6.16.7-edge-rockchip64.  It seems to be using "rkvdec" vpu driver.

Whereas kernel-6.16.8-1 Chromium could not use or does not seem to be able use "rkvdec" for vpu hardware acceleration.

```
[alarm@opi5plus ~]$ uname -a
Linux opi5plus 6.16.7-edge-rockchip64 #1

[alarm@opi5plus ~]$ echo "v4l2 video devices"
v4l2_devices=`ls -1 /dev/video*`
for d in ${v4l2_devices}; do
  echo "--------------------------------------------------------------------------------"
  echo "Device: ${d}:"
  v4l2-ctl -d ${d} -D --list-formats-ext --list-formats-out-ext
  echo "--------------------------------------------------------------------------------"
  echo " "
done
v4l2 video devices
--------------------------------------------------------------------------------
Device: /dev/video0:
Driver Info:
        Driver name      : rockchip-rga
        Card type        : rockchip-rga
        Bus info         : platform:rga
        Driver version   : 6.16.7
        Capabilities     : 0x84204000
                Video Memory-to-Memory Multiplanar
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04204000
                Video Memory-to-Memory Multiplanar
                Streaming
                Extended Pix Format
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture Multiplanar

        [0]: 'BA24' (32-bit ARGB 8-8-8-8)
        [1]: 'AR24' (32-bit BGRA 8-8-8-8)
        [2]: 'XR24' (32-bit BGRX 8-8-8-8)
        [3]: 'RGB3' (24-bit RGB 8-8-8)
        [4]: 'BGR3' (24-bit BGR 8-8-8)
        [5]: 'AR12' (16-bit ARGB 4-4-4-4)
        [6]: 'AR15' (16-bit ARGB 1-5-5-5)
        [7]: 'RGBP' (16-bit RGB 5-6-5)
        [8]: 'NV21' (Y/VU 4:2:0)
        [9]: 'NV61' (Y/VU 4:2:2)
        [10]: 'NV12' (Y/UV 4:2:0)
        [11]: 'NM12' (Y/UV 4:2:0 (N-C))
        [12]: 'NV16' (Y/UV 4:2:2)
        [13]: 'YU12' (Planar YUV 4:2:0)
        [14]: '422P' (Planar YUV 4:2:2)
        [15]: 'YV12' (Planar YVU 4:2:0)
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'BA24' (32-bit ARGB 8-8-8-8)
        [1]: 'AR24' (32-bit BGRA 8-8-8-8)
        [2]: 'XR24' (32-bit BGRX 8-8-8-8)
        [3]: 'RGB3' (24-bit RGB 8-8-8)
        [4]: 'BGR3' (24-bit BGR 8-8-8)
        [5]: 'AR12' (16-bit ARGB 4-4-4-4)
        [6]: 'AR15' (16-bit ARGB 1-5-5-5)
        [7]: 'RGBP' (16-bit RGB 5-6-5)
        [8]: 'NV21' (Y/VU 4:2:0)
        [9]: 'NV61' (Y/VU 4:2:2)
[... 169 lines cut ...]
        Driver version   : 6.16.7
Interface Info:
        ID               : 0x0300000c
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : rkvdec-source
        Function         : V4L2 I/O
        Pad 0x01000002   : 0: Source
          Link 0x02000008: to remote pad 0x1000004 of entity 'rkvdec-proc' (Video Decoder): Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture Multiplanar

        [0]: 'NV12' (Y/UV 4:2:0)
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'S265' (HEVC Parsed Slice Data, compressed)
                Size: Continuous 1x1 - 65472x65472
        [1]: 'S264' (H.264 Parsed Slice Data, compressed)
                Size: Continuous 1x1 - 65520x65520
--------------------------------------------------------------------------------
 
[alarm@opi5plus ~]$ 
```

closed 2025-10-01 by warpme